ARMmbed/mbed-os #3904: I2C constructor failure on NUCLEO_F746ZG when using PinName PA_8

mray19027, 2017-03-07:
### Description

Creating an I2C instance using SCL pin PA_8 causes a PinMap mis-match

SCL pin PA_8 is stated to be a valid SCL pin as shown below:
```
//*** I2C ***
const PinMap PinMap_I2C_SDA[] = {
    // {PB_7,  I2C_1, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C1)}, // (used by LED2)
       {PB_9,  I2C_1, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C1)},
       {PB_11, I2C_2, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C2)},
       {PC_9,  I2C_3, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C3)},
       {PD_13, I2C_4, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C4)},
       {PF_0,  I2C_2, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C2)},
       {PF_15, I2C_4, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C4)},
       {NC,    NC,    0}
};

const PinMap PinMap_I2C_SCL[] = {
    {PA_8,  I2C_3, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C3)},
    {PB_6,  I2C_1, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C1)},
    {PB_8,  I2C_1, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C1)},
    {PB_10, I2C_2, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C2)},
    {PD_12, I2C_4, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C4)},
    {PF_1,  I2C_2, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C2)},
    {PF_14, I2C_4, STM_PIN_DATA(STM_MODE_AF_OD, GPIO_NOPULL, GPIO_AF4_I2C4)},
    {NC,    NC,    0}
};
```

Code to replicate error:
```
#include "mbed.h"
#include "cmsis.h"
#include "pinmap.h"
#include "PeripheralPins.h"

int main() {
    PinName pin_sda = PB_9;
    PinName pin_scl = PA_8;
    printf("SDA pin %d, SCL pin %d\r\n", pin_sda, pin_scl);

    I2C i2c(pin_sda, pin_scl);

    printf("This will never run because the I2C constructor will fail for this SDA/SCL pairing\r\n");
}
```

### Details

**Target**
NUCLEO_F746ZG 

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 

**mbed-cli version:** 
1.0.0

**meed-os sha:**
e435a07d9

@maclobdell @BlackstoneEngineering

LMESTM, 2017-03-07:
@mray19027 When muxing the 2 pins of an HW block, you need the 2 pins to belong to the same block instance. PA_8 has muxing capability for SDA line of _I2C_3_, whereas PB_9 has muxing capability for SCL line of _I2C_1_, so this will fail for sure ...

mray19027, 2017-03-07:
@LMESTM Thank you! Closing the issue

closed 2017-03-07 by mray19027