ARMmbed/mbed-os #3905: Uvisor importer: Build fails with GNU make 4.2.1

rjpdasilva, 2017-03-07:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
Any supported by `uvisor`.

**Toolchain:**
GCC_ARM (required for building `uvisor`)

**Toolchain version:**
`6.3.0`

**mbed-cli version:** 
`1.0.0`

**mbed-os sha:**
`b2726470f Merge pull request #3880 from adustm/fix_can2_only`

**Expected behavior**
Building `uvisor` succeeds, independently of the `make` version in use.

**Actual behavior**
Building fails with `GNU Make 4.2.1`.
Building succeeds with `GNU Make 3.81`.

**Steps to reproduce**
* Using `GNU Make 4.2.1`.
* Fetch a clean `mbed-os` repo.
* Build `uvisor`with:
`make -C mbed-os/features/FEATURE_UVISOR/importer`
* Note the build failure.

Build failure:
```
# Copying licenses
cp TARGET_IGNORE/uvisor/LICENSE* ../targets/TARGET_UVISOR_SUPPORTED
make: *** No rule to make target 'TARGET_M3', needed by 'publish'.  Stop.
make: Leaving directory '/home/rps/scm/rps/iot/arm_mbed/rps_tests/mbed-os-example-uvisor/mbed-os/features/FEATURE_UVISOR/importer'
```

NOTE: After the error, doing `make` again does succeed. 

**Preliminary Analysis**
Check the `features/FEATURE_UVISOR/importer/Makefile` file:
```make
TARGET_M%: $(TARGET_SUPPORTED)/*/*/*_cortex_m%*.a
	@printf "#\n# Copying $@ files...\n"
	mkdir $(foreach file,$^,$(dir $(file))$@)
	$(foreach file,$^,mv $(file) $(dir $(file))$@/lib$(notdir $(file));)

publish: rsync TARGET_M3 TARGET_M4
```

It seems that the target `TARGET_M%` depends on outputs created by the `rsync` target. These outputs are the `$(TARGET_SUPPORTED)/*/*/*_cortex_m%*.a` files.
 However, there's no explicit dependency set between them in the `Makefile`.

The `publish` target needs `rsync` and both `TARGET_M3` and `TARGET_M4`, but there's nothing preventing `make` from trying to do, e.g., `TARGET_M3` before `rsync`.

Additionally, the wildcard/pattern expansion should be performed before actually creating the targets, so for the `TARGET_M%` rule, that will happen at a time where the required files (`$(TARGET_SUPPORTED)/*/*/*_cortex_m%*.a`) were not created yet, causing no rule definition for the targets M3 nor M4. I believe it is this that is causing the problem.
This also explains why the second `make` does succeed: at that 2nd run, the required files are already there. But if you run a 3rd time, it will fail again, a 4th time will succeed again and so on.

I don't know exactly why this doesn't happen with `GNU Make 3.81`. The only explanation is that it is only doing the expansion when the `publish` target is being executed. 

The missing explicit dependencies between `TARGET_M%` and `rsync` is also a problem, because even with `GNU Make 3.81`, the build can fail when we use `make -jN` (with N > 1).

I have a tentative solution for both the problems described. Will submit it through a pull request.

0xc0170, 2017-03-08:
cc @Patater @AlessandroA - please review (there is also fix referenced above, sent to this repository)

0xc0170, 2017-03-13:
Fixed via referenced PR above

closed 2017-03-13 by 0xc0170