ARMmbed/mbed-os #3907: NRF51_DK have FAIL on CI test case 'tests-api-i2c'

TsungtaWu, 2017-03-08:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NRF51_DK

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3

**mbed-cli version:** 
1.0.0

**Expected behavior**
 In CI test, K64F is able to pass all test case 'tests-api-i2c'

**Actual behavior**
NRF51_DK have three FAIL as shown below
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-i2c               | I2C -  EEProm 2nd WR 10  Bytes                             | 1      | 0      | OK     | 0.05               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-i2c               | I2C -  EEProm 2nd WR 100 Bytes                             | 1      | 0      | OK     | 0.07               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-i2c               | I2C -  EEProm 2nd WR 2 Bytes                               | 1      | 0      | OK     | 0.07               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-i2c               | I2C -  EEProm 2nd WR Single Byte                           | 0      | 2      | FAIL   | 0.21               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-i2c               | I2C -  EEProm WR 10  Bytes                                 | 1      | 0      | OK     | 0.05               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-i2c               | I2C -  EEProm WR 100 Bytes                                 | 1      | 0      | OK     | 0.07               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-i2c               | I2C -  EEProm WR 2 Bytes                                   | 0      | 3      | FAIL   | 0.24               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-i2c               | I2C -  EEProm WR Single Byte                               | 0      | 2      | FAIL   | 0.21               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-i2c               | I2C -  Instantiation of I2C Object                         | 1      | 0      | OK     | 0.06               |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-api-i2c               | I2C -  LM75B Temperature Read                              | 1      | 0      | OK     | 0.07               |

**Steps to reproduce**
 CI shield V1.0.0 mount on NRF51_DK platform
 goto ci-test-shield folder, then type
 $ mbed test -m NRF51_DK -t GCC_ARM -n tests-api-i2c* --app-config mbed_app.json

0xc0170, 2017-03-08:
One thing - did you test the version that is checked in ci test repository or updated to different version? There was a fix integrated recently, we can compare if you have already tested it or not. Here's the fix: https://github.com/ARMmbed/mbed-os/pull/3812 (5.4.0 will include this fix)

cc @anangl @nvlsianpu

0xc0170, 2017-03-13:
@TsungtaWu Can you please review the above answer? I wonder if there's still i2c failures with ci shield or it's fixed .

btw, we are going to update CI test shield repository for the latest release (you can do it now manually, i2c should build and run).

TsungtaWu, 2017-03-14:
@0xc0170 Sorry for the late reply.
I just try to test the i2c with 5.4.0 but I got a compile error on my side
        collect2.exe: error: ld returned 1 exit status
[api-i2c_log.txt](https://github.com/ARMmbed/mbed-os/files/840227/api-i2c_log.txt)

I type following commands 
$ cd mbed-os
$ git checkout mbed-os-5.4
$ mbed update
$ cd ..
$  mbed test -n tests-api-i2c* --app-config .\mbed_app.json 
Is anything I miss?

nvlsianpu, 2017-03-14:
It fail for me as well. I tried to compile this test for K64F - with the same errors.

0xc0170, 2017-03-14:
@TsungtaWu this is known-issue (reported in the ci-test shield), will be fixed. To overcome this, remove SDFilesystem that is causing the linker errors you shared above.

Once you do , you should get this output:

```
+--------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
| target       | platform_name | test suite    | test case                          | passed | failed | result | elapsed_time (sec) |
+--------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
| NRF51_DK-ARM | NRF51_DK      | tests-api-i2c | I2C -  EEProm 2nd WR 10  Bytes     | 1      | 0      | OK     | 0.06               |
| NRF51_DK-ARM | NRF51_DK      | tests-api-i2c | I2C -  EEProm 2nd WR 100 Bytes     | 1      | 0      | OK     | 0.09               |
| NRF51_DK-ARM | NRF51_DK      | tests-api-i2c | I2C -  EEProm 2nd WR 2 Bytes       | 1      | 0      | OK     | 0.06               |
| NRF51_DK-ARM | NRF51_DK      | tests-api-i2c | I2C -  EEProm 2nd WR Single Byte   | 1      | 0      | OK     | 0.07               |
| NRF51_DK-ARM | NRF51_DK      | tests-api-i2c | I2C -  EEProm WR 10  Bytes         | 1      | 0      | OK     | 0.06               |
| NRF51_DK-ARM | NRF51_DK      | tests-api-i2c | I2C -  EEProm WR 100 Bytes         | 1      | 0      | OK     | 0.08               |
| NRF51_DK-ARM | NRF51_DK      | tests-api-i2c | I2C -  EEProm WR 2 Bytes           | 1      | 0      | OK     | 0.06               |
| NRF51_DK-ARM | NRF51_DK      | tests-api-i2c | I2C -  EEProm WR Single Byte       | 1      | 0      | OK     | 0.06               |
| NRF51_DK-ARM | NRF51_DK      | tests-api-i2c | I2C -  Instantiation of I2C Object | 1      | 0      | OK     | 0.07               |
| NRF51_DK-ARM | NRF51_DK      | tests-api-i2c | I2C -  LM75B Temperature Read      | 1      | 0      | OK     | 0.05               |
+--------------+---------------+---------------+------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 10 OK
```

closed 2017-03-14 by 0xc0170

TsungtaWu, 2017-03-15:
@0xc0170 Issue is fixed manually on my side as well.