ARMmbed/mbed-os #3910: Filesystem changes made for mbed-os-5.4 have broken mbed 2 compilation

adbridge, 2017-03-08:
### Description

Running the mbed-2-release job on the mbed-os-5.4-rc2 release gives 40 target compilation failures (compared with 0 for 5.3.6)

Example failure: 
http://10.118.12.43:8080/job/mbed-2-build-library/1251/testReport/build.ARCH_GPRS.IAR/MBED/mbed_SDK/

```
Compile [ 15.6%]: FileBase.cpp
[Error] FileLike.h@48,48: [Pe020]: identifier "size_t" is undefined
[Error] FileLike.h@56,14: [Pe020]: identifier "size_t" is undefined
[Error] FileLike.h@103,0: [Pe020]: identifier "size_t" is undefined

      virtual ssize_t read(void *buffer, size_t len) = 0;
                                         ^
"C:\Jenkins\workspace\mbed-2-build-library\BUILD\mbed\drivers\FileLike.h",48  Error[Pe020]: identifier "size_t" is undefined

      virtual ssize_t write(const void *buffer, size_t len) = 0;
                                                ^
"C:\Jenkins\workspace\mbed-2-build-library\BUILD\mbed\drivers\FileLike.h",56  Error[Pe020]: identifier "size_t" is undefined

      virtual size_t size() = 0;
              ^
"C:\Jenkins\workspace\mbed-2-build-library\BUILD\mbed\drivers\FileLike.h",103  Error[Pe020]: identifier "size_t" is undefined
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os/issues/3854
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F|??

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

adbridge, 2017-03-08:
@geky @simonqhughes

0xc0170, 2017-03-08:
See the fix : https://github.com/ARMmbed/mbed-os/pull/3864

0xc0170, 2017-03-13:
Fixed via the referenced pull request

closed 2017-03-13 by 0xc0170