ARMmbed/mbed-os #3916: Cannot link mbed os with GCC_ARM and mbed cli on Windows undefined reference to `main`

adrien59cadri, 2017-03-09:
Hello all

I have a clean environment, with mbed-cli installed, GCC_ARM toolchain and target NUCLEO_F401RE.
Compilation goes well but link of main fails:

`Compile [ 97.1%]: port_api.c
Compile [ 97.4%]: rtc_api.c
Compile [ 97.8%]: i2c_api.c
Compile [ 98.2%]: pwmout_api.c
Compile [ 98.5%]: sleep.c
Compile [ 98.9%]: us_ticker_16b.c
Compile [ 99.3%]: stm_spi_api.c
Compile [ 99.6%]: us_ticker_32b.c
Compile [100.0%]: test_env.cpp
Link: mbed-os
./BUILD/NUCLEO_F401RE/GCC_ARM/platform/mbed_retarget.o: In function `__wrap_main':
mbed_retarget.cpp:(.text.__wrap_main+0xa): undefined reference to `main'
collect2.exe: error: ld returned 1 exit status
[ERROR] ./BUILD/NUCLEO_F401RE/GCC_ARM/platform/mbed_retarget.o: In function `__wrap_main':
mbed_retarget.cpp:(.text.__wrap_main+0xa): undefined reference to `main'
collect2.exe: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Users\adrie\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_F401RE --source . --build .\BUILD\NUCLEO_F401RE\GCC_ARM" in "C:\Users\adrie\mbed-os"
---
`

Would you have any idea, i guess an include must be missing somewhere but don't really see where.

0xc0170, 2017-03-10:
what are you building ? is there main function defined -> building an application ?

adrien59cadri, 2017-03-10:
Hello I'm very sorry, I was trying to build the os, i thought it was generating a lib but actually it just wants to buld an app and means I should create the main.
Sorry again, and thanks for your feedback

closed 2017-03-10 by adrien59cadri