ARMmbed/mbed-os #3919: mbed-os export to gnuarmeclipse doesn't build

bulislaw, 2017-03-10:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
EFM32GG_STJ3700

**Toolchain:**
GCC_ARM

I'm trying to export some tests as a mbed OS application to debug them in eclipse:
```
cp TESTS/mbedmicro-rtos-mbed/threads/* 
mbed export -m EFM32GG_STK3700 -i gnuarmeclipse
```
The build works using mbed cli but fails using eclipse:

```
10:35:43 **** Clean-only build of configuration Debug for project mbed-os ****
make -j3 mbedclean 
rm -rf  ./tools/data/rpc/class.o  ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_acmp.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_adc.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_aes.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_assert.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_burtc.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_cmu.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_core.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_cryotimer.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_crypto.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_dac.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_dbg.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_dma.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_ebi.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_emu.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_gpcrc.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_gpio.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_i2c.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_idac.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_int.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_lcd.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_ldma.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_lesense.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_letimer.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_leuart.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_mpu.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_msc.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_opamp.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_pcnt.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_prs.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_rmu.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_rtc.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_rtcc.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_system.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_timer.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_usart.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_vcmp.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_wdog.o  ./targets/TARGET_Silicon_Labs/TARGET_EFM32/common/mbed_overrides.o  ./targets/TARGET_Silicon_Labs/TARGET_EFM32/TARGET_EFM32GG/device/TARGET_1024K/TOOLCHAIN_GCC_ARM/startup_efm32gg.o  ./targets/TARGET_Silicon_Labs/TARGET_EFM32/TARGET_EFM32GG/device/system_efm32gg.o  ./targets/TARGET_Silicon_Labs/TARGET_EFM32/TARGET_EFM32GG/PeripheralPins.o  ./targets/TARGET_Silicon_Labs/TARGET_EFM32/analogin_api.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/analogout_api.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/dma_api.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/gpio_api.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/gpio_irq_api.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/i2c_api.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/lp_ticker.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/pinmap.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/pinmap_function.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/port_api.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/pwmout_api.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/rtc_api.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/serial_api.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/sleep.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/spi_api.o ./targets/TARGET_Silicon_Labs/TARGET_EFM32/us_ticker.o  ./rtos/rtx2/TARGET_CORTEX_M/TARGET_M3/TOOLCHAIN_GCC/irq_cm3.o  ./rtos/rtx2/TARGET_CORTEX_M/RTX_Config.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_delay.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_evflags.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_evr.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_kernel.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_lib.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_memory.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_mempool.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_msgqueue.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_mutex.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_semaphore.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_system.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_thread.o ./rtos/rtx2/TARGET_CORTEX_M/rtx_timer.o  ./rtos/Mutex.o ./rtos/Semaphore.o ./rtos/Thread.o ./rtos/mbed_boot.o ./rtos/mbed_rtx_handlers.o ./rtos/rtos_idle.o  ./platform/CallChain.o ./platform/mbed_alloc_wrappers.o ./platform/mbed_application.o ./platform/mbed_assert.o ./platform/mbed_board.o ./platform/mbed_critical.o ./platform/mbed_error.o ./platform/mbed_interface.o ./platform/mbed_mem_trace.o ./platform/mbed_retarget.o ./platform/mbed_rtc_time.o ./platform/mbed_semihost_api.o ./platform/mbed_stats.o ./platform/mbed_wait_api_no_rtos.o ./platform/mbed_wait_api_rtos.o  ./hal/mbed_gpio.o ./hal/mbed_lp_ticker_api.o ./hal/mbed_pinmap_common.o ./hal/mbed_ticker_api.o ./hal/mbed_us_ticker_api.o  ./features/netsocket/NetworkInterface.o ./features/netsocket/NetworkStack.o ./features/netsocket/Socket.o ./features/netsocket/SocketAddress.o ./features/netsocket/TCPServer.o ./features/netsocket/TCPSocket.o ./features/netsocket/UDPSocket.o ./features/netsocket/WiFiAccessPoint.o ./features/netsocket/nsapi_dns.o  ./features/mbedtls/src/aes.o ./features/mbedtls/src/aesni.o ./features/mbedtls/src/arc4.o ./features/mbedtls/src/asn1parse.o ./features/mbedtls/src/asn1write.o ./features/mbedtls/src/base64.o ./features/mbedtls/src/bignum.o ./features/mbedtls/src/blowfish.o ./features/mbedtls/src/camellia.o ./features/mbedtls/src/ccm.o ./features/mbedtls/src/certs.o ./features/mbedtls/src/cipher.o ./features/mbedtls/src/cipher_wrap.o ./features/mbedtls/src/cmac.o ./features/mbedtls/src/ctr_drbg.o ./features/mbedtls/src/debug.o ./features/mbedtls/src/des.o ./features/mbedtls/src/dhm.o ./features/mbedtls/src/ecdh.o ./features/mbedtls/src/ecdsa.o ./features/mbedtls/src/ecjpake.o ./features/mbedtls/src/ecp.o ./features/mbedtls/src/ecp_curves.o ./features/mbedtls/src/entropy.o ./features/mbedtls/src/entropy_poll.o ./features/mbedtls/src/error.o ./features/mbedtls/src/gcm.o ./features/mbedtls/src/havege.o ./features/mbedtls/src/hmac_drbg.o ./features/mbedtls/src/md.o ./features/mbedtls/src/md2.o ./features/mbedtls/src/md4.o ./features/mbedtls/src/md5.o ./features/mbedtls/src/md_wrap.o ./features/mbedtls/src/memory_buffer_alloc.o ./features/mbedtls/src/net_sockets.o ./features/mbedtls/src/oid.o ./features/mbedtls/src/padlock.o ./features/mbedtls/src/pem.o ./features/mbedtls/src/pk.o ./features/mbedtls/src/pk_wrap.o ./features/mbedtls/src/pkcs11.o ./features/mbedtls/src/pkcs12.o ./features/mbedtls/src/pkcs5.o ./features/mbedtls/src/pkparse.o ./features/mbedtls/src/pkwrite.o ./features/mbedtls/src/platform.o ./features/mbedtls/src/ripemd160.o ./features/mbedtls/src/rsa.o ./features/mbedtls/src/sha1.o ./features/mbedtls/src/sha256.o ./features/mbedtls/src/sha512.o ./features/mbedtls/src/ssl_cache.o ./features/mbedtls/src/ssl_ciphersuites.o ./features/mbedtls/src/ssl_cli.o ./features/mbedtls/src/ssl_cookie.o ./features/mbedtls/src/ssl_srv.o ./features/mbedtls/src/ssl_ticket.o ./features/mbedtls/src/ssl_tls.o ./features/mbedtls/src/threading.o ./features/mbedtls/src/timing.o ./features/mbedtls/src/version.o ./features/mbedtls/src/version_features.o ./features/mbedtls/src/x509.o ./features/mbedtls/src/x509_create.o ./features/mbedtls/src/x509_crl.o ./features/mbedtls/src/x509_crt.o ./features/mbedtls/src/x509_csr.o ./features/mbedtls/src/x509write_crt.o ./features/mbedtls/src/x509write_csr.o ./features/mbedtls/src/xtea.o  ./features/mbedtls/platform/src/mbed_trng.o  ./features/frameworks/utest/source/unity_handler.o ./features/frameworks/utest/source/utest_case.o ./features/frameworks/utest/source/utest_default_handlers.o ./features/frameworks/utest/source/utest_greentea_handlers.o ./features/frameworks/utest/source/utest_harness.o ./features/frameworks/utest/source/utest_shim.o ./features/frameworks/utest/source/utest_stack_trace.o ./features/frameworks/utest/s [... cut ...]
 class Rpc{{name}} : public RPC {
                 ^
../tools/data/rpc/class.cpp:1:19: error: expected unqualified-id before ':' token
 class Rpc{{name}} : public RPC {
                   ^
make: *** [tools/data/rpc/class.o] Error 1
make: *** Waiting for unfinished jobs....
Finished building: ../targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_acmp.c
Finished building: ../targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/src/em_adc.c
 
 

10:35:47 Build Finished (took 1s.116ms)
```

@theotherjimmy

theotherjimmy, 2017-03-10:
your `cp` command is mising a destination.

theotherjimmy, 2017-03-10:
Which eclipse? we have a few :P

bulislaw, 2017-03-13:
I executed the copy correctly (the main.c and some headers are in the sources). Now also fixed in bug report.

haishengliu, 2017-05-17:
make all 
Building file: ../ARMmbed_5p3_r897_PLC/tools/data/rpc/class.cpp
Invoking: Cross ARM GNU C++ Compiler
arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -O0 -fmessage-length=0 -fsigned-char -ffunction-
tools/data/rpc/class.cpp:1:11: error: expected unqualified-id before '{' token
 class Rpc{{name}} : public RPC {
           ^
tools/data/rpc/class.cpp:1:17: error: expected ';' after class definition
 class Rpc{{name}} : public RPC {
                 ^
tools/data/rpc/class.cpp:1:19: error: expected unqualified-id before ':' token
 class Rpc{{name}} : public RPC {
                   ^
make: *** tools/data/rpc/subdir.mk:20: tools/data/rpc/class.o] Error 1

18:20:05 Build Finished (took 26s.247ms)

theotherjimmy, 2017-05-17:
Huh. the tools directory contains an mbedignore with a `*` in it. That file should not be included.

haishengliu, 2017-05-18:
so, should i delete this tools folder?

theotherjimmy, 2017-05-18:
The fact that the files are being included is a bug.

JojoS62, 2017-07-18:
I can reproduce this behaviour for OS2. For OS5, the tools folder is excluded (by the gnuarmeclipse plugin?). For OS2 projects, the folder is in '.temp' and this folder is not shown in the project tree. But the CDT scanner finds this folder and the cpp source file. So this genrates the error.

Manual fix: Exclude the .tools subdir from the resources in the eclipse project settings:
Resource/Resources Filters , Add Filter
[image]

[image]

This creates an entry in the .project file for excluding this folder.

theotherjimmy, 2017-07-18:
@JojoS62 I'll look into how the exclude search is failing WRT mbed 2. You are correct that the tools folder should be excluded from builds.

JojoS62, 2017-07-18:
I have tried to create a PR with the modified template file, but the PR contains all my commits for other stuff. So I need to read first more about the git stuff. Anyway, you should see my modification in the closed PR.
But maybe @ilg-ul has also another way for configuring exclusions in his plugin?

JojoS62, 2017-07-18:
@theotherjimmy found something more: there is a dictionary excluded_folders for this purpose. It is the same for mbed2 and 5 in https://github.com/ARMmbed/mbed-os/blob/master/tools/export/gnuarmeclipse/__init__.py#L468
```
        # print 'excludings'
        self.excluded_folders = ['BUILD']
        self.recurse_excludings(self.source_tree)
```

theotherjimmy, 2017-07-18:
Yep. (that's a list, not a dict, BTW). 

I was suggesting that `recursive_excludings` should add `.temp` to that list.

theotherjimmy, 2017-07-18:
@JojoS62 I convert the tools folder of mbed OS to a release by applying a smallish patch, so most things should be the same.

JojoS62, 2017-07-18:
yes, I agree its better than modifying the template.

theotherjimmy, 2017-07-18:
```diff
diff --git a/tools/build_api.py b/tools/build_api.py
index c9e288dae..bbc1e0ae7 100644
--- a/tools/build_api.py
+++ b/tools/build_api.py
@@ -34,12 +34,13 @@ from tools.paths import MBED_CMSIS_PATH, MBED_TARGETS_PATH, MBED_LIBRARIES,\
     MBED_HEADER, MBED_DRIVERS, MBED_PLATFORM, MBED_HAL, MBED_CONFIG_FILE,\
     MBED_LIBRARIES_DRIVERS, MBED_LIBRARIES_PLATFORM, MBED_LIBRARIES_HAL,\
     BUILD_DIR
-from tools.targets import TARGET_NAMES, TARGET_MAP
+from tools.targets import TARGET_NAMES, TARGET_MAP, set_targets_json_location
 from tools.libraries import Library
-from tools.toolchains import TOOLCHAIN_CLASSES
+from tools.toolchains import TOOLCHAIN_CLASSES, mbedToolchain
 from jinja2 import FileSystemLoader
 from jinja2.environment import Environment
 from tools.config import Config
+from tools.build_profiles import find_build_profile, get_toolchain_profile, find_targets_json
 
 RELEASE_VERSIONS = ['2', '5']
 
@@ -487,6 +488,45 @@ def build_project(src_paths, build_path, target, toolchain_name,
         rmtree(build_path)
     mkdir(build_path)
 
+    ###################################
+    # mbed Classic/2.0/libary support #
+
+    # Find build system profile
+    profile = None
+    targets_json = None
+    for path in src_paths:
+        profile = find_build_profile(path) or profile
+        if profile:
+            targets_json = join(dirname(abspath(__file__)), 'legacy_targets.json')
+        else:
+            targets_json = find_targets_json(path) or targets_json
+
+    # Apply targets.json to active targets
+    if targets_json:
+        if verbose:
+            print("Using targets from %s" % targets_json)
+        set_targets_json_location(targets_json)
+
+    # Apply profile to toolchains
+    if profile:
+        def init_hook(self):
+            profile_data = get_toolchain_profile(self.name, profile)
+            if not profile_data:
+                return
+            if verbose:
+                self.info("Using toolchain %s profile %s" % (self.name, profile))
+
+            for k,v in profile_data.items():
+                if self.flags.has_key(k):
+                    self.flags[k] = v
+                else:
+                    setattr(self, k, v)
+
+        mbedToolchain.init = init_hook
+
+    # mbed Classic/2.0/libary support #
+    ###################################
+
     toolchain = prepare_toolchain(
         src_paths, build_path, target, toolchain_name, macros=macros,
         clean=clean, jobs=jobs, notify=notify, silent=silent, verbose=verbose,
diff --git a/tools/config/__init__.py b/tools/config/__init__.py
index e6d888fda..80fb5005e 100644
--- a/tools/config/__init__.py
[... 78 lines cut ...]
+        mbedToolchain.init = init_hook
+
+    # mbed Classic/2.0/libary support #
+    ###################################
+
     return export_project(src, project_dir, target, ide, name=name,
                           macros=macros, libraries_paths=lib, zip_proj=zip_name,
                           build_profile=build_profile, silent=silent,
diff --git a/tools/targets/__init__.py b/tools/targets/__init__.py
index 9d0217423..1789feb54 100644
--- a/tools/targets/__init__.py
+++ b/tools/targets/__init__.py
@@ -120,7 +120,7 @@ class Target(namedtuple("Target", "name json_data resolution_order resolution_or
 
     # Default location of the 'targets.json' file
     __targets_json_location_default = os.path.join(
-        os.path.dirname(os.path.abspath(__file__)), '..', '..', 'targets', 'targets.json')
+        os.path.dirname(os.path.abspath(__file__)), '..', 'latest_targets.json')
 
     # Current/new location of the 'targets.json' file
     __targets_json_location = None

```

That's the diff (with a few new files removed).

theotherjimmy, 2017-07-18:
I'm going to look real quick to see if the .temp folder is being missed for some silly reason.

ilg-ul, 2017-07-18:
I'm not sure I understand the use case.

generally the exporter should run in a fresh folder, where there are no other folders generated by other exporters, or temporary folders from other builds. otherwise the python code used by the exporter will pick up files which are not actually needed and I could not find a rule to ignore them.

and also I'm confused by the way you want to use the exporter, i.e. to run various make commands on the result. the exporter generates managed projects, which you open with Eclipse and build inside Eclipse.

theotherjimmy, 2017-07-18:
Actually, We are just debugging using the exporter, as you describe, for mbed 2 projects. In mbed 2 projects, the tools folder is located at `.temp/tools` and that appears to be indexed by GNU ARM Eclipse after export.

ilg-ul, 2017-07-18:
you can patch the exporter to exclude `.temp`, if needed.

theotherjimmy, 2017-07-18:
@ilg-ul I'm pretty sure that the mbed build system does not include the `.temp` dir, so your exclusion mechanism should pick up on this and add an exclude rule for it. I'm going to look into it to see if there is a bug in one of the steps leading up to export or in the exporter.

ilg-ul, 2017-07-19:
feel free to add any exclusions as you think necessary.

JojoS62, 2017-07-19:
from my understanding:
- compute_exclusions reads the whole project tree and marks sources as 'used' and other as 'not used'
- the iteration skips folders that start with a '.' in the name. Eclipse does not show such directorys in the tree, there must be an option for this in the workspace settings?
- because the '.temp' is not parsed, the sourcefiles in this subdir get not excluded
- for eclipse CDT, it includes by default everything that it finds. Exceptions where not to search can be added by the excluded_folder setting or the resource filter that I described.

This is why I like the zip option where only the necessary sources are picked and added to the zip. Eclipse can import this as well and many operations work much faster because it hat not to analyse the large target tree. A zipped mbed2 library base is about 2 MB, the complete mbed2 library repo is already about 1,2 GB. Using the sources when exporting from mbed-os5 is also fine, but compilation takes much longer (there is always a reason for build-all).
So for the exporter it would be fine to have both options: --library for using the precompiled lib and --zip for creating a single target extract. When possible with the VCS information included.

theotherjimmy, 2017-07-19:
@JojoS62 We could add a `-z/--zip` command line option, in addition to fixing the normal mbed 2 export. 

> This is why I like the zip option where only the necessary sources are picked and added to the zip. 

Sounds like a feature request to me. This would have another advantage: you could generate the same thing that the online tools do by just passing `-z/--zip`

theotherjimmy, 2017-07-19:
@ilg-ul It sounds like we need to add the exclusion rule `.*` for parody with the build system, or drop the skipping of `.` prefixed files from the exclude generator.

theotherjimmy, 2017-07-19:
@JojoS62 @ilg-ul It looks like the following diff does what we want with no discernible performance impact:
```diff
diff --git i/tools/export/gnuarmeclipse/__init__.py w/tools/export/gnuarmeclipse/__init__.py
index d6de69f4b..28123caaa 100644
--- i/tools/export/gnuarmeclipse/__init__.py
+++ w/tools/export/gnuarmeclipse/__init__.py
@@ -437,14 +437,6 @@ class GNUARMEclipse(Exporter):
                 # to accomodate Windows weirdness.
                 parts = root.split(os.sep)
 
-                # Ignore paths that include parts starting with dot.
-                skip = False
-                for part in parts:
-                    if part.startswith('.'):
-                        skip = True
-                        break
-                if skip:
-                    continue
 
                 # Further process only leaf paths, (that do not have
                 # sub-folders).
```

theotherjimmy, 2017-07-19:
I have a `.git` and `.settings` folder in my project, and it looks like they are being added to the exclude list:
before
```
Create a GNU ARM Eclipse C++ managed project
Project name: mbed-os-example-fat-filesystem
Target: K64F
Toolchain: GCC_ARM
Source folders: 42, with 73 exclusions
```
after
```
Create a GNU ARM Eclipse C++ managed project
Project name: mbed-os-example-fat-filesystem
Target: K64F
Toolchain: GCC_ARM
Source folders: 42, with 75 exclusions
```

JojoS62, 2017-07-19:
Yes. I hope this has no impact on the git plugin, I can test this later.

JojoS62, 2017-07-19:
They are in the exclude list and this is ok because there is no c code for the project, but the git plugin should still see these directories.

theotherjimmy, 2017-07-19:
@JojoS62 I'm a bit confused. Are you saying that the `.git` folder contains no C code? or that the project contains no C code?

theotherjimmy, 2017-07-19:
I'm testing with an mbed 2 project, and it's not excluding `.temp` :(

ilg-ul, 2017-07-19:
my suggestion is to explicitly ignore `.git` and `.settings`, I guess Eclipse does not parse them for source code anyway and you might save a few cycles on each export.

theotherjimmy, 2017-07-19:
I'm guessing you meant to add it to the array of static ignores on this line:
```python
        self.excluded_folders = ['BUILD']
```
I don't particularly like that. It seems brittle. If I add more dot-files with C code in them we will have to continue adding more entries to the static ignores.

I'm trying to make the exclude generating code generate those excludes dynamically so that we remove a whole class of potential incompatibilities between this exporter and `mbed compile`

ilg-ul, 2017-07-19:
I don't see why this is brittle, processing `.git` by the exporter is a definite waste of time, since Eclipse does not parse it anyway. each extra file processed by the exporter adds a new node in the internal tree, and the `.git` folder has **lots** of files.

theotherjimmy, 2017-07-19:
@ilg-ul The difference in processing time seems to be on the order of ~10ms. 

I have a much less brittle proposal: have the Resource object track directories that were ignored during the resource scan. This should not add a significant overhead, and it would be very useful for these exporters.

theotherjimmy, 2017-07-19:
@ilg-ul @JojoS62 The patch that has the resource scan do our work for us looks good. It drops ~100ms off the export time for gnuarmeclipse for mbed-os exports, and does not affect `mbed compile` times.

ilg-ul, 2017-07-19:
I have no idea what your proposal mean.

whatever you think, processing `.git` in the exporter is pointless, regardless how much processing time it adds. 

> It seems brittle. If I add more dot-files with C code in them we will have to continue adding more entries to the static ignores

frankly, I have no idea where this concern came from.

we are talking here about removing the code to filter out the dot folders. that code was added there mainly for `.git` and `.settings` (and possibly the .DS_Store) . without that code, the unwanted folders were explicitly added to `.cproject` as exclusions; probably this is not harmfull, but it is definitely not usual or needed.

and I think you do not have to add any static ignore for those two folders.

I don't mind what solution you use, but I would prefer not to generate silly `.cproject` files.

theotherjimmy, 2017-07-19:
@ilg-ul My proposed solution is to modify `scan_resources` to collect directories that it ignored while scanning. Then, in the exporters, simply use this list instead of repeating the scan. Alternatively, see PR #4783.

theotherjimmy, 2017-07-19:
>>It seems brittle. If I add more dot-files with C code in them we will have to continue adding more entries to the static ignores
>frankly, I have no idea where this concern came from.

The mbed 2 projects include the tools within a directory called `.temp`. The tools, for whatever reason, contain a template file with a `.c` extension. Including this file breaks builds. We need to ignore this directory.

I'd also rather solve this once and for all. But agreed, who, in their right mind, would add a directory starting with `.` and have `.c` or similar files in it!

ilg-ul, 2017-07-19:
I'm not familiar with the detailed logic behind your code, but, as I said, as long as it does not generate unwanted exclusions in `.cproject`, it should be fine.

theotherjimmy, 2017-07-19:
Yeah. That's fine. It generates all of the exclusions used by `mbed compile` by collecting the directory removals at their source. 

TL;DR/Don't care: The exclusions are now exactly the same as `mbed compile`.

JojoS62, 2017-07-19:
@theotherjimmy I don't konw ecactly how your solution looks like but it sounds good. I could also reproduce that not skipping the '.*' directrories does not work.
Another thought: it would be nice if the directory names for the tools in mbed2 and 5 would be the same. But I think thats history and more difficult to get.

About the zip option:
When I force 'zip_file = True' (in project.py main()), a project.zip is created in the root of the project. This can be imported in eclipse and it works. The zipfiles also doesn't have the .temp problem because it is not included. Just 2 MB with the pure single target project. This is also the behaviour for the exporters before the big mbed5 rework.

theotherjimmy, 2017-07-19:
>Another thought: it would be nice if the directory names for the tools in mbed2 and 5 would be the same. But I think thats history and more difficult to get.

Yeah, that ship has sailed.

> When I force 'zip_file = True' (in project.py main()), a project.zip is created in the root of the project.

This is exactly what the website produces as well, if it succeeds. 

I thinks this thread boils down to a bug and a feature request:
 * the `.temp` directory was not being ignored (fix in #4783)
 * Allow website and mbed classic style exports with a `-z/--zip` option.

I'll get to the second feature request tomorrow.

JojoS62, 2017-07-19:
Great! 
And I learned more important things in this puzzle :-)

JojoS62, 2017-07-19:
just one more thing (bug?):

> This is exactly what the website produces as well, if it succeeds.

I got also an error with creating the zipfile, a file not found error for toos/export/.mbed. There I thought I had deleted it accidently, but I cannot see it in the repo:
https://developer.mbed.org/users/mbed_official/code/mbed-sdk-tools/file/399953da035d/export

Either this is not needed or missing in the export dir, maybe this is a reason for the export error on the website?

theotherjimmy, 2017-07-19:
Oh, that was a miss-copy. Let me get that for you.

theotherjimmy, 2017-07-19:
Naw, the website uses the tools from a patched version of the mbed OS release from github. That's not the export problem on the website. I have tracked the website export problem down. We're verifying now and going to get it deployed ASAP.

theotherjimmy, 2017-07-19:
I pushed the `.mbed` file up for you. Sorry about that.

theotherjimmy, 2018-06-25:
#4783 fixed this bug.

closed 2018-06-25 by theotherjimmy