ARMmbed/mbed-os #3923: ATSAML21 blinky build failed

vargham, 2017-03-10:
### Description
- Type: Bug
- Priority: Blocker
Tried to experiment with https://developer.mbed.org/teams/Atmel/code/Atmel_Blinky/
Could not build project.
Error: Cannot open source input file "device.h": No such file or directory in "extras/mbed_e1686b8d5b90/platform/platform.h", Line: 27, Col: 21
---------------------------------------------------------------
## Bug

**Target**
Atmel ATSAML21-XPRO

**Toolchain:**
Online IDE

**Expected behavior**
Build, upload, blink LED.

**Actual behavior**
Build failes, device.h not found.

**Steps to reproduce**
Import https://developer.mbed.org/teams/Atmel/code/Atmel_Blinky/ select Atmel ATSAML21-XPRO target and try to build.

0xc0170, 2017-03-13:
cc @Parthasarathy @akhilpanayamparambil

0xc0170, 2017-03-13:
@theotherjimmy - can you please check that this target is well-defined in the online IDE?

Parthasarathy, 2017-03-13:
Hi Martin,

    I will spend some time this week to verify this issue.

Thanks,
Partha

or mute
the thread

akhilpanayamparambil, 2017-03-15:
@0xc0170 @vargham 
We have not added modules for L21 target in the yotta build. So I am not quite sure if it supports L21 targets for the blinky example.

In the list of dependencies in for mbed does not list L21. (includes R21 and D21 XPROs)
https://developer.mbed.org/users/mbed_official/code/mbed/

0xc0170, 2017-03-15:
Here's the platform: https://developer.mbed.org/platforms/SAML21-XPRO/

 This is taken from targets.json file:
```
    "SAML21J18A": {
        "inherits": ["Target"],
        "core": "Cortex-M0+",
        "macros": ["__SAML21J18A__", "I2C_MASTER_CALLBACK_MODE=true", "EXTINT_CALLBACK_MODE=true", "USART_CALLBACK_MODE=true", "TC_ASYNC=true"],
        "extra_labels": ["Atmel", "SAM_CortexM0P", "SAML21"],
        "supported_toolchains": ["GCC_ARM", "ARM", "uARM"],
        "device_has": ["ANALOGIN", "ANALOGOUT", "I2C", "I2CSLAVE", "I2C_ASYNCH", "INTERRUPTIN", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT", "RTC", "SERIAL", "SERIAL_ASYNCH", "SERIAL_FC", "SLEEP", "SPI", "SPISLAVE", "SPI_ASYNCH"],
        "device_name": "ATSAML21J18A"
    },
```

Is this target definition for this platform? I don't see the release version defined there. Should this target be enabled?

> In the list of dependencies in for mbed does not list L21. (includes R21 and D21 XPROs)
https://developer.mbed.org/users/mbed_official/code/mbed/

Seems like L21 should not be in the platforms, and neither the code is not available for this platform?

cc @screamerbg

closed 2017-10-27 by ChiefBureaucraticOfficer