ARMmbed/mbed-os #3928: Build Intel Hex for multiple ROM support

ccli8, 2017-03-13:
### Description
- Type: Enhancement
- Priority: Major

----------------------------------------------------------------
## Enhancement
We (Nuvoton) are adding new chip port (NUMAKER_PFM_M487) for mbed OS. In this chip, there are two ROM blocks:
```
ROM1 (512 KB): 0x0000_0000 – 0x0007_FFFF
ROM2 (32 MB): 0x0800_0000 – 0x09FF_FFFF
```
The two ROM blocks are non-contiguous. As part code is placed in ROM2, the generated binary file would get very large with `GCC_ARM` and `IAR` toolchains. There are two raw binary files generated with `ARM` toolchain. So instead of raw binary format, we expect just one Intel Hex file could be generated directly from ELF file and the Intel Hex file could flash to mbed board directly for Greentea test. Could you guild how to support this?

0xc0170, 2017-03-13:
> As part code is placed in ROM2, the generated binary file would get very large with GCC_ARM and IAR toolchains.

Can you elaborate?

> So instead of raw binary format, we expect just one Intel Hex file could be generated directly from ELF file and the Intel Hex file could flash to mbed board directly for Greentea test. Could you guild how to support this?

Is this a feature request for online IDE? Or also mbed-cli to produce intel hex file?

ccli8, 2017-03-14:
@0xc0170 My test for conversion of ELF to binary for two ROM images
1. ARM (mbed-os/tools/toolchains/arm.py > binary): Two separate binary files generated
    ```
    fromelf --bin -o bin elf
    ```
1. GCC_ARM (mbed-os/tools/toolchains/gcc.py > binary): One large, combined binary file generated due to padding
    ```
    objcopy -O binary elf bin
    ```
1. IAR (mbed-os/tools/toolchains/iar.py > binary): One large, combined binary file generated due to padding
    ```
    ielftool --bin elf bin
   ```
To support multi-ROM platform like us, it seems it is necessary to support Intel Hex format. Please support it both on mbed-cli and online IDE.

0xc0170, 2017-03-14:
Thanks @ccli8 , its more clear now.

> To support multi-ROM platform like us, it seems it is necessary to support Intel Hex format. Please support it both on mbed-cli and online IDE.

It is supported there, for instance for nordic devices. Therefore your target can use the same functionality. Does your target support hex drag-and-drop flashing?

cc @theotherjimmy @sg- @c1728p9

ccli8, 2017-03-14:
> It is supported there, for instance for nordic devices. Therefore your target can use the same functionality. 

@0xc0170 If I don't get wrong, nordic's and other targets' hex support need post-processing for generated binary file from ELF with python code (`mbed-os/nu-porting/mbed-os/tools/targets.py`). But for two-ROM case as above, the generated binary file would get large due to non-contiguous. It is expected that we can instruct tool to generate hex from ELF directly without intermediary binary. Per our test, the three toolchains (ARM/GCC_ARM/IAR) above all can support conversion from ELF to hex.

> Does your target support hex drag-and-drop flashing?

Yes (NUMAKER_PFM_NUC472/NUMAKER_PFM_M453).

0xc0170, 2017-03-14:
There was an addition for another target to handle similar use case, https://github.com/ARMmbed/mbed-os/pull/3340 . We are waiting for GCC to support it (should be added soon).

ccli8, 2017-06-29:
Verified OK on mbed-os-5.5.0. mbed CLI and Online Compiler both can output Intel HEX file as OUTPUT_EXT is set to hex.

closed 2017-06-29 by ccli8