ARMmbed/mbed-os #3932: STM32 headers defining __packed?

kjbracey-arm, 2017-03-14:
It appears that many or all STM32 HAL headers define `__packed` as `__attribute__((__packed__))`, and this define is then exposed to any source file including mbed headers.

That seems like it can break ArmCC compilation - lwIP for example tries to use the ArmCC syntax `__packed struct`, but that gets replaced by `__attribute__((__packed__)) struct`, which then generates a warning about an ineffective attribute - ArmCC does understand that GCC attribute, but the attribute needs to go after the declaration, not before.

You see a lot of these warnings on any STM32 ArmCC build with lwIP in.

So I'm not sure how lwIP works on any STM32 platform. Are all its __packed keywords actually redundant? Or are the misplaced attributes working despite the warning?

It seems to me that the STM headers shouldn't be overriding built-in compiler keywords.

Just changing the test in the STM headers from `ifdef __GNUC__` to `if defined __GNUC__ && !defined __CC_ARM` might be enough - they're possibly being understandably confused by ArmCC defining `__GNUC__`.

Thoughts welcome...

0xc0170, 2017-03-14:
> It seems to me that the STM headers shouldn't be overriding built-in compiler keywords.

+1

I checked F4 targets, this packed macro is located in ``stm32f4xx_hal_def.h``: https://github.com/ARMmbed/mbed-os/blob/f59b5f8b75223937137240da0fbc3556de318ec3/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_def.h#L127-L134

@bcostm @adustm @LMESTM @jeromecoutant Please have a look

LMESTM, 2017-03-14:
We'll report this to our HAL team and let you know. 
Note that I would have been confused as well by ArmCC defining __GNUC__ - and that was my first question when reading the first part of the issue description. 

Would you explain why this is the case, also whether this is new (e.g. a CMSIS5 update ?)

0xc0170, 2017-03-14:
> Would you explain why this is the case, also whether this is new (e.g. a CMSIS5 update ?)

some of the code in the code base requires it thus has been enabled since I can remember. Even if you disable this flag, the code should be corrected as ARMCC provides this option, and this use case should be handled.

LMESTM, 2017-03-14:
Sure we'll fix and the proposal to use _if defined __GNUC__ && !defined __CC_ARM_ can even be fine in the short term and I can send it. I just wanted to be able to give some more background to the team here, I just understood now this is seems to be related to using the ARMCC --gnu  command line option, is it right ?

0xc0170, 2017-03-14:
> I just understood now this is seems to be related to using the ARMCC --gnu command line option, is it right ?

That's correct ``--gnu``` option.  This enables the GNU compiler extensions that the ARM compiler supports. more info: https://developer.arm.com/docs/dui0472/l/compiler-command-line-options/-gnu

LMESTM, 2017-04-27:
@0xc0170  @kjbracey-arm  - PR sent to correct the warning issue

bcostm, 2017-05-18:
ST_INTERNAL_REF 11317

LMESTM, 2017-06-05:
@kjbracey-arm @0xc0170 PR with fix has been merged - may you close the issue ?

closed 2017-06-05 by 0xc0170