ARMmbed/mbed-os #3935: Teensy 3.5 Support

pctj101, 2017-03-15:
- Type: Enhancement | Question
- Priority: Minor

----------------------------------------------------------------
## Enhancement
** Support Teensy 3.5 / 3.6 **

It would be lovely to see mbed support for the Teensy 3.5.  

Why? The Teensy is in a very convenient form factor with USB.  Makes me really want to use it.

I see that @geky added the Teensy 3.1 support.  There are some magical files here that I don't the source of (CMSIS items) but looks like it's somewhat humanly possible.

-----------------------------------------------------------------

## Question

I'm trying to find a little time to dig through some SDK files directly related to the CPU.  Is that all that is necessary?  Or is it more along the lines of pin mapping? Timing / etc?  

I figure there's probably some reason that boards aren't exactly flying on to this list.  Perhaps it's more work than I realize.

0xc0170, 2017-03-15:
It was ported by @Sissors and @star297 as I recall. Those files were moved thus the history is not preserved (at least as it is viewable here on github).

star297, 2017-03-15:
See my comment here:

https://developer.mbed.org/questions/74731/Teensy-3536/

I not convinced its really worth the effort for Mbed considering the the Power, BLE and IOT limitations of both the development board and MCU itself.
Maybe a good platform for Arduino however there may be better alternative targets on Mbed.

I can't see any problems if NXP wish to write the Mbed library for this target, don't forget it would not be true Mbed drag 'n' drop programming for this particular board, the same as with the Teensy 3.1

Sissors, 2017-03-16:
I think processing power and form factor wise it would be a nice board to have added. But for me Teensy 3.1 cost too much effort to get added to mbed. And I don't know if the one for 3.5/3.6 also would require the KSDK driver library that I really dislike :P.

pctj101, 2017-03-16:
@star297 If I'm asking in the wrong place, let me know. I'm still new to mbed.

I was never really much a fan of the drag-drop file into a drive style of deployment.  I was reading about the CMSIS-DAP interface which could program the chip which I thought was pretty cool since it would not just program, but allow debugging. 

When you say that it isn't "true mbed drag 'n' drop", which protocol can't be supported?  And is it related to the HalfKay bootloader in the Teensy?  

@Sissors I equally see a lot of files related to adding in the teensy.  Is there something that makes the teensy especially a lot of work? (same MCU as a FRDM board)  Or is it just not popular enough?  I'm not looking for justification, just asking out of general curiosity.

Anyways, I just happened to have a teensy, so if there's a better fit for what I want to do, economically, it's just cheaper for me to buy a different board and let everyone concentrate on what makes the greatest sense.

Sissors, 2017-03-16:
I only now see one works on a different K64 version than the FRDM-K64 board, but still K64. That should be fairly easy to add, K66 might be a larger effort. 

So the Teensy 3.5 should be reasonable quick to add. But without support from mbed it will then take ages before it would be added to the online compiler.

pctj101, 2017-03-17:
Ah, okay.  Yeah from where I am, I'm planning to integrate the result into PlatformIO, so my usage pattern does not have any dependency on mbed online (just couldn't stand programming in a textarea)

And right!  The Teensy 3.5 is probably more "closely related" and conveniently what I have.  (Nevermind the Teensy 3.6 hahaha)

If something magically appears and works "off-line" I'm definitely using it!

closed 2017-10-27 by ChiefBureaucraticOfficer