ARMmbed/mbed-os #3938: Export - scan only root folder, only config file in the workspace

0xc0170, 2017-03-16:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
any

**Toolchain:**
any

**mbed-cli version:** 
1.0.0

**meed-os sha:**
master (sha: cc58a7f ) or latest mbed-os-5.4

**Expected behavior**
Scan all directories within root, and ahaving all files in the workspace view.

**Actual behavior**

**Steps to reproduce**
Fetch mbed-os blinky example. Deploy and export to iar or uvision.

Here are my outputs:

mbed-os 5.3:
```
test-rtos\mbed-os (master)                                                                                                           
λ git co mbed-os-5.3                                                                                                                      
Branch mbed-os-5.3 set up to track remote branch mbed-os-5.3 from origin.                                                                 
Switched to a new branch 'mbed-os-5.3'                                                                                                    
                                                                                                                                          
test-rtos\mbed-os (mbed-os-5.3)                                                                                                      
λ cd ../                                                                                                                                  
                                                                                                                                          
test-rtos (master)                                                                                                                   
λ mbed export -m NUCLEO_F401RE -i iar                                                                                                     
C:\Python27\lib\site-packages\fuzzywuzzy\fuzz.py:35: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to re
ve this warning                                                                                                                           
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')                              
Scan: .                                                                                                                                   
Scan: FEATURE_BLE                                                                                                                         
Scan: FEATURE_COMMON_PAL                                                                                                                  
Scan: FEATURE_LWIP                                                                                                                        
Scan: FEATURE_UVISOR                                                                                                                      
Scan: FEATURE_ETHERNET_HOST                                                                                                               
Scan: FEATURE_LOWPAN_BORDER_ROUTER                                                                                                        
Scan: FEATURE_LOWPAN_HOST                                                                                                                 
Scan: FEATURE_LOWPAN_ROUTER                                                                                                               
Scan: FEATURE_NANOSTACK                                                                                                                   
Scan: FEATURE_NANOSTACK_FULL                                                                                                              
Scan: FEATURE_THREAD_BORDER_ROUTER                                                                                                        
Scan: FEATURE_THREAD_END_DEVICE                                                                                                           
Scan: FEATURE_THREAD_ROUTER                                                                                                               
Scan: FEATURE_STORAGE                                                                                                                     
```

Using latest master:

```
test-rtos\mbed-os (master)
λ git logg -1
*   cc58a7f - (HEAD, origin/master, origin/HEAD, master) Merge pull request #3927 from TsungtaWu/master (2 days ago) <Anna Bridge>
|\

test-rtos\mbed-os (master)
λ cd ../

test-rtos (master)
λ mbed export -m NUCLEO_F401RE -i iar
Scan: .
Traceback (most recent call last):
  File "test-rtos\mbed-os\tools\project.py", line 248, in <module>
    main()
  File "test-rtos\mbed-os\tools\project.py", line 244, in main
    build_profile=profile)
  File "test-rtos\mbed-os\tools\project.py", line 94, in export
    build_profile=build_profile, silent=silent)
  File "test-rtos\mbed-os\tools\project_api.py", line 222, in export_project
    macros=macros)
  File "test-rtos\mbed-os\tools\project_api.py", line 89, in generate_project_files
    exporter.generate()
  File "test-rtos\mbed-os\tools\export\iar\__init__.py", line 122, in generate
    'linker_script': self.format_file(self.resources.linker_script),
  File "test-rtos\mbed-os\tools\export\iar\__init__.py", line 89, in format_file
    return join('$PROJ_DIR$',file)
  File "C:\Python27\lib\ntpath.py", line 66, in join
    p_drive, p_path = splitdrive(p)
  File "C:\Python27\lib\ntpath.py", line 114, in splitdrive
    if len(p) > 1:
TypeError: object of type 'NoneType' has no len()
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u test-rtos\mbed-os\tools\project.py -i iar -m NUCLEO_F401RE --source ." in "test-rtos"
---

IAR fails to even export. uvision exports, but contains only config header file
```

@theotherjimmy Please have a look.

0xc0170, 2017-03-16:
Just found this, the same report but for example - https://github.com/ARMmbed/mbed-os-example-tls/issues/67

The issue has a fix https://github.com/ARMmbed/mbed-os/pull/3924 !

theotherjimmy, 2017-04-26:
@0xc0170 That fix was merged some time back, can we close this issue?

closed 2017-04-27 by 0xc0170