ARMmbed/mbed-os #3940: Compiling error with GCC toolchain (K64F)

RdaZhongyao, 2017-03-16:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Related issue: `#compiler`
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
K64F

**Environment:**
Windows/msys2/python2.7.11/pip9.0.1
install other tools : "pip install -r mbed-os-5.4.1/requirements.txt"

**Toolchain:**
GCC_ARM 

**Toolchain version:**
4.9

**mbed-cli version:** 
(`mbed --version`) : 1.0.0

**meed-os sha:**
(`git log -n1 --oneline`) : f4864dc6429e1ff5474111d4e0f6bee36a759b1c (Release ver: 5.4.1)

**Expected behavior**
Compile the code with GCC_ARM

**Actual behavior**
Compiling error:
```
[Fatal Error] AnalogIn.cpp@17,30: drivers/AnalogIn.h: No such file or directory
[DEBUG] Return: 1
[DEBUG] Output: ./drivers\AnalogIn.cpp:17:30: fatal error: drivers/AnalogIn.h: No such file or directory
[DEBUG] Output:  #include "drivers/AnalogIn.h"
[DEBUG] Output:                               ^
[DEBUG] Output: compilation terminated.
Traceback (most recent call last):
  File "E:\mbed\mbed-os-5.4.1\tools\make.py", line 285, in <module>
    toolchain))
  File "E:\mbed\mbed-os-5.4.1\tools\build_api.py", line 512, in build_project
    objects = toolchain.compile_sources(resources, resources.inc_dirs)
  File "E:\mbed\mbed-os-5.4.1\tools\toolchains\__init__.py", line 824, in compile_sources
    return self.compile_queue(queue, objects)
  File "E:\mbed\mbed-os-5.4.1\tools\toolchains\__init__.py", line 887, in compile_queue
    raise ToolException(err)
ToolException: ./drivers\AnalogIn.cpp:17:30: fatal error: drivers/AnalogIn.h: No such file or directory
 #include "drivers/AnalogIn.h"
                              ^
compilation terminated.
```

**Steps to reproduce**
```
$ cd mbed-os-5.4.1
$ mbed compile -m K64F -t GCC_ARM --source TESTS/integration/basic --source ./ 
```

RdaZhongyao, 2017-03-17:
## Updated
While I change my command:
```
$ mbed compile -m K64F -t GCC_ARM --source TESTS/integration/basic --source .
```
That's OK. So the directory `--source .` can be parsed by GCC tools, `--source ./` is not.

In addition, there is no problem for ARM compiler.

0xc0170, 2017-03-17:
@RdaZhongyao Thanks for the update, its more clear now.

@theotherjimmy Please have a look

theotherjimmy, 2017-03-20:
I cannot reproduce the issue.

``` bash
$  mbed compile -t gcc_arm -m k64f --source ./ --source TESTS/integration/basic
```
produces
```
Compile [ 99.7%]: main.cpp
Compile [100.0%]: test_env.cpp
Link: mbed
Elf2Bin: mbed
+-----------+-------+-------+-------+
| Module    | .text | .data |  .bss |
+-----------+-------+-------+-------+
| Fill      |   112 |     4 |  2401 |
| Misc      | 44417 |  2556 |  8143 |
| Subtotals | 44529 |  2560 | 10544 |
+-----------+-------+-------+-------+
Allocated Heap: 24576 bytes
Allocated Stack: unknown
Total Static RAM memory (data + bss): 13104 bytes
Total RAM memory (data + bss + heap + stack): 37680 bytes
Total Flash memory (text + data + misc): 48129 bytes

Image: ./BUILD/k64f/gcc_arm/mbed.bin
```

RdaZhongyao, 2017-03-21:
Maybe the differrent environments produce differrent results. I use Windows/Python2.7.11/GCC4.9.
```
mbed compile -t GCC_ARM -m K64F --source TESTS/integration/basic --source ./
```
produces
```
Building project basic (K64F, GCC_ARM)
Scan: basic
Scan:
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
Scan: env
Scan: mbed
Compile [  0.3%]: AnalogIn.cpp
[Fatal Error] AnalogIn.cpp@17,30: drivers/AnalogIn.h: No such file or directory
[ERROR] ./drivers\AnalogIn.cpp:17:30: fatal error: drivers/AnalogIn.h: No such file or directory
 #include "drivers/AnalogIn.h"
                              ^
compilation terminated.

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u E:\mbed\mbed-os-5.4.1\tools\make.py -t GCC_ARM -m K64F --source TESTS/integration/basic --source ./ --build .\BUILD\K64F\GCC_ARM" in "E:\mbed\mbed-os-5.4.1"
---
```
After this, a file named `.includes_628003686ef9dbaf11132ab7ae6dbd2e.txt` is generated in dir `BUILD/K64F/GCC_ARM/`, read the file:
```
$ cat BUILD/K64F/GCC_ARM/.includes_628003686ef9dbaf11132ab7ae6dbd2e.txt
-I./ -I./cmsis -I./cmsis/TOOLCHAIN_GCC -I./drivers -I./events -I./events/equeue ......
```
Then I modify the file from `-I./` to `-I.`
```
$ cat BUILD/K64F/GCC_ARM/.includes_628003686ef9dbaf11132ab7ae6dbd2e.txt
-I. -I./cmsis -I./cmsis/TOOLCHAIN_GCC -I./drivers -I./events -I./events/equeue ......
```
Run compiling command again
```
mbed compile -t GCC_ARM -m K64F --source TESTS/integration/basic --source ./
```
produces
```
Compile [ 99.7%]: main.cpp
Compile [100.0%]: test_env.cpp
Link: basic
Elf2Bin: basic
+-----------+-------+-------+-------+
| Module    | .text | .data |  .bss |
+-----------+-------+-------+-------+
| Fill      |   112 |     4 |  2145 |
| Misc      | 46959 |  2292 |  8143 |
| Subtotals | 47071 |  2296 | 10288 |
+-----------+-------+-------+-------+
Allocated Heap: 24576 bytes
Allocated Stack: unknown
Total Static RAM memory (data + bss): 12584 bytes
Total RAM memory (data + bss + heap + stack): 37160 bytes
Total Flash memory (text + data + misc): 50407 bytes

Image: .\BUILD\K64F\GCC_ARM\basic.bin
```

@theotherjimmy  anything can help?

0xc0170, 2017-03-21:
I was able to reproduce it (windows machine) the same as above:``mbed compile -t GCC_ARM -m K64F --source TESTS/integration/basic --source ./`` this fails, if I change the last one to ``.`` only, compiles.

theotherjimmy, 2017-03-21:
thanks @RdaZhongyao for the debug information. That will help trace this back to a solution.

theotherjimmy, 2017-03-22:
This definetly looks like a bug. I'll see what I can do about it.

PR #4170 (2017-04-19): Fix include paths for Windows builds
# Descritpion

Resolves #3940

# How to reproduce
The issue mentioned above can me sumarized as:

This does not work:
```bash
mbed compile --source ./
```

This works:
```bash
mbed compile --source .
```

Both should work.

## Why does this fail?

The include directories passed to `--source` are added as-is to the list
of include directories. GCC, and possibly other compilers, on Windows 
does not like the path `./` and will ignore it when searching for header
files.

# New behavior

Remove trailing `/`s before appending include paths to when scanning 
reso [... cut]
closed 2017-04-19 by sg-