ARMmbed/mbed-os #3941: osThreadSuspend and osThreadResume

Wazalogic, 2017-03-16:
I need to suspend and resume threads. Keil RTX specifies the osThreadSuspend(osThreadId) and osThreadResume(osThreadId) functions, but unfortunately they are not implemented by this OS.

I am aware that these functions are dangerous to use. I really need them otherwise I cannot continue with my project using mbed.

Is there a way to implement these functions with SVC? Any examples?

bulislaw, 2017-03-16:
'osThreadSuspend(osThreadId)' and  'osThreadResume(osThreadId) ' are new to CMSIS5/RTX2, currently we are in a process of upgrade right now, you'll need to stick to CMSIS4/RTX1 APIs for now. http://www.keil.com/pack/doc/CMSIS/RTOS/html/group__CMSIS__RTOS__ThreadMgmt.html
I don't think RTX1 defines anything like the mentioned functions, possibly you can manipulate thread priorities to not having it scheduled, but that seems a bit far fetched and I don't know details of the scheduler implementation. Maybe worth asking Keil guys here https://github.com/ARM-software/CMSIS Maybe you can change your application, so that the thread you may need to suspend, can wait on signal/semaphore instead?

Wazalogic, 2017-03-16:
Thanks. I need these functions. Is there a FreeRTOS port for that works with mbed-os 5?

bulislaw, 2017-03-16:
I'm not aware about anything like that.

0xc0170, 2017-03-17:
@Wazalogic You can try our development branch with RTX2 (https://github.com/ARMmbed/mbed-os/tree/feature_cmsis5) However note, this is not yet stable - thus development branch. And yet no user api in the RTOS API we provide there.

If you look at ``Thread`` it has methods ``start`` and ``terminate`` - not satisfactory for your app?

bulislaw, 2017-03-23:
i'll close this ticket for now. Please reopen it if you have some further questions.

closed 2017-03-23 by bulislaw