ARMmbed/mbed-os #3952: Buffer overflow in channel_free in nrf_drv_gpiote.c

c1728p9, 2017-03-16:
## Bug

**Target**
NRF5x family

**Toolchain:**
GCC_ARM|ARM|IAR

A buffer overflow can occur in the gpiote driver [here](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5/sdk/drivers_nrf/gpiote/nrf_drv_gpiote.c#L160),  leading to undefined behavior.

Callstack at the time of failure is below. The parameter channel_id was set to 255 which lead to the overflow.
```
channel_free() at nrf_drv_gpiote.c:162 
nrf_drv_gpiote_in_uninit() at nrf_drv_gpiote.c:474 
gpiote_pin_uninit() at gpio_api.c:100
gpio_mode() at gpio_api.c:157
_gpio_init_in() at mbed_gpio.c:23
gpio_init_in_ex() at mbed_gpio.c:42
gpio_init_in() at mbed_gpio.c:38
mbed::DigitalIn::DigitalIn() at DigitalIn.h:60
mbed::BusIn::BusIn() at BusIn.cpp:27
busin_define_test() at BusInOut.cpp:28
utest::v1::Harness::run_next_case() at utest_harness.cpp:338
utest_us_ticker_run() at utest_shim.cpp:123
```

**Steps to reproduce**

Checkout master of the [CI test shield repo](https://github.com/ARMmbed/ci-test-shield) and Update mbed-os to 5.2.0 
```
>> mbed import https://github.com/ARMmbed/ci-test-shield
>> cd mbed-os
>> mbed update mbed-os-5.2.0
>> cd ../
```

Add bounds check to channel_free
```
static void channel_free(uint8_t channel_id)
{
+    const uint32_t elements = sizeof(m_cb.handlers) / sizeof(m_cb.handlers[0]);
+    if (channel_id >= elements) {
+        error("Out of bounds error would have occurred");
+    }
    m_cb.handlers[channel_id] = FORBIDDEN_HANDLER_ADDRESS;
    if (channel_id >= NUMBER_OF_GPIO_TE)
    {
        m_cb.port_handlers_pins[channel_id-NUMBER_OF_GPIO_TE] = (int8_t)PIN_NOT_USED;
    }
}
```

Build and run Bus In Out  test
```
mbed test -t GCC_ARM -m nrf51_dk -n tests-api-businout --app-config .\mbed_app.json -v
```

c1728p9, 2017-03-16:
CC: @senthilr @BlackstoneEngineering

pan-, 2017-03-16:
@nvlsianpu @anangl Could you look at this bug ?

closed 2017-10-27 by ChiefBureaucraticOfficer