ARMmbed/mbed-os #3958: Invalid character `#' in linker script (STM32F4 with ARM_GCC)

vargham, 2017-03-16:
### Description
- Type: Bug
- Priority: Minor
Tried to build STM32F4 project with ARM_GCC, but could not build because of linker error.

---------------------------------------------------------------
## Bug
STM32F429xI.ld from line 13
`#if !defined(MBED_APP_START)
  #define MBED_APP_START 0x08000000
#endif

#if !defined(MBED_APP_SIZE)
  #define MBED_APP_SIZE 2048k
#endif

/* Specify the memory areas */
MEMORY
{
    VECTORS (rx) : ORIGIN = MBED_APP_START, LENGTH = 0x400
    FLASH (rx)   : ORIGIN = MBED_APP_START + 0x400, LENGTH = MBED_APP_SIZE - 0x400
    CCM (rwx)    : ORIGIN = 0x10000000, LENGTH = 64K
    RAM (rwx)    : ORIGIN = 0x20000000, LENGTH = 192k
}`
**Target**
STM32F4

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi 5.4.1

**Expected behavior**
Exported project compile and link without errors.

**Actual behavior**
STM32F429xI.ld:13: ignoring invalid character `#' in expression
STM32F429xI.ld:13: syntax error

**Steps to reproduce**
Create project for STM32F4. Export to offline IDE (ARM_GCC) and build.

0xc0170, 2017-03-17:
cc @bcostm @adustm @LMESTM @jeromecoutant

adustm, 2017-03-17:
@c1728p9 @theotherjimmy I think you are already aware ?
I think this comes from PR #3749 ?

@vargham in #3749 there is a suggestion of workaround for your issue.
Let us know if it works.
Kind regards

vargham, 2017-03-17:
I use the EmBitz exporter. My workaround: I changed the linker script back to the previous version without ifdefs.

0xc0170, 2017-03-17:
> @c1728p9 @theotherjimmy I think you are already aware ?
I think this comes from PR #3749. Could you check this, please ?

Yes , this is intended. I did not earlier spot it's about macros in the linker script.

> I use the EmBitz exporter. My workaround: I changed the linker script back to the previous version without ifdefs.

OK I now see what is the failure here. The linker scripts are now required to be preprocessed. Therefore EmBitz exporter should be updated to do this step.

This bug is then for the exporter, not the target.

adustm, 2017-03-17:
Thank you Martin @0xc0170

toyokomi, 2017-05-20:
Hello, so what is the solution. I got problem, as it is desribed above. I exported project from mbed Compiler into SW4STM32, but after build a received message:
`path.../STM32F429xI.ld:13: ignoring invalid character # in expression`
`path.../STM32F429xI.ld:13: syntax error`
Thanks in advance!

0xc0170, 2017-05-25:
@theotherjimmy Is there anything we can do?

amq, 2017-06-18:
I have the same problem with EFM32PG12 on 5.5.0 rc2.

bcostm, 2017-06-26:
Launched today a SW4STM32 export with mbed online (on DISCO_L476VG) and still the same error with the #if/#define not recognized in the ld file.

From #3749 :
> You are correct to point out this deficiency in the exporter. The problem is that it does not run the C preprocessor on the linker script before passing it to the linker.

Any idea when this preprocessing will be fixed. Or can you give some hints ?

MarceloSalazar, 2017-08-01:
> Any idea when this preprocessing will be fixed. Or can you give some hints ?

I understand the pre-processing of linker scripts is already working. Could you please re-test and confirm?

theotherjimmy, 2017-08-01:
@MarceloSalazar this issue is still open and correct as there have not been any updates to the SW4STM32 exporter.

AchmadFathoni, 2018-01-20:
I found a workaround based on succesfull earlier version of mbed-os. first delete this part of code in STM32F446XE.ld
```
#if !defined(MBED_APP_START)
  #define MBED_APP_START 0x08000000
#endif

#if !defined(MBED_APP_SIZE)
  #define MBED_APP_SIZE 512K
#endif
```
And replace 
```
MEMORY
{
  FLASH (rx)   : ORIGIN = MBED_APP_START, LENGTH = MBED_APP_SIZE
  RAM (rwx)  : ORIGIN = 0x200001C4, LENGTH = 128k - 0x1C4
}
```
with
```
MEMORY
{ 
  FLASH (rx) : ORIGIN = 0x08000000, LENGTH = 512K
  RAM (rwx)  : ORIGIN = 0x200001C4, LENGTH = 128k - 0x1C4
}
```

closed 2018-01-23 by 0xc0170