ARMmbed/mbed-os #3959: IAR Compilation Failed With LPC1768 In Mbed OS 5.4.0

pranav3672, 2017-03-17:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Major

---------------------------------------------------------------
## Bug  :- Compilation failed with IAR.

Target  :- LPC1768
Toolchain :- IAR
Toolchain version : 7.4
mbed OS version :- 5.4.0
DAPLink version :- 243

**Expected behavior**
LPC1768 should be compile with IAR.

**Actual behavior**
Failed to compile.

**Steps to reproduce**
1. Open Git Bash
2. Go to tools directory of Mbed OS 5.4.0 package.
3. Type below command
    python build.py -m LPC1768 -t IAR
[image]

----------------------------------------------------------------
## Enhancement
Include the "stddef.h" file in any of the target HAL directory.
Note:- Please include "stddef.h" in header file only.
For Example:- 
Open objects.h file :- ..\mbed-os-mbed-os-5.4.0\targets\TARGET_NXP\TARGET_LPC176X\objects.h
Add #include <stddef.h> (Line No. 18 )
Then Compile.
-----------------------------------------------------------------

0xc0170, 2017-03-17:
You are using build script that is for mbed 2. This however is fixed on master (there was this bug reported more than a week ago), and I believe even released for the latest mbed lib. Please retest and reopen if thats not the case.

Latest master:

```
Completed in: (27.68)s

Build successes:
  * IAR::LPC1768
```

0xc0170, 2017-03-17:
To be certain, branch mbed-os-5.4 contains also that fix as I wrote earlier.

```
Completed in: (16.72)s

Build successes:
  * IAR::LPC1768

git branch
  master
* mbed-os-5.4
```

closed 2017-03-17 by 0xc0170

theotherjimmy, 2017-03-20:
@0xc0170 Could you tell @pranav3672 Which script to run instead?

0xc0170, 2017-03-21:
> @0xc0170 Could you tell @pranav3672 Which script to run instead?

It's fine to use build.py at the moment