ARMmbed/mbed-os #3963: EFM32 Pearl Gecko export broken

chrissnow, 2017-03-17:
- Type: Bug 
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
EFM32PG

**Toolchain:**
ARM

**Toolchain version:**

**mbed-cli version:** 
1.0.0

Exporting to uVision or Simplicity Studio online or by CLI is broken.
`
mbed export -i  uvision5 -m efm32pg_stk3401
Scan: .
`
and then returns to console

`
mbed export -i simplicityv3 -m efm32pg_stk3401
Scan: .
Traceback (most recent call last):
  File "C:\Code\Ekkosense\EkkoDisplayEFM\mbed-os\tools\project.py", line 248, in <module>
    main()
  File "C:\Code\Ekkosense\EkkoDisplayEFM\mbed-os\tools\project.py", line 244, in main
    build_profile=profile)
  File "C:\Code\Ekkosense\EkkoDisplayEFM\mbed-os\tools\project.py", line 94, in export
    build_profile=build_profile, silent=silent)
  File "C:\Code\Ekkosense\EkkoDisplayEFM\mbed-os\tools\project_api.py", line 222, in export_project
    macros=macros)
  File "C:\Code\Ekkosense\EkkoDisplayEFM\mbed-os\tools\project_api.py", line 89, in generate_project_files
    exporter.generate()
  File "C:\Code\Ekkosense\EkkoDisplayEFM\mbed-os\tools\export\simplicity\__init__.py", line 158, in generate
    self.check_and_add_path(split(self.resources.linker_script)[0])
  File "c:\python27\lib\ntpath.py", line 180, in split
    d, p = splitdrive(p)
  File "c:\python27\lib\ntpath.py", line 115, in splitdrive
    if len(p) > 1:
TypeError: object of type 'NoneType' has no len()
[mbed] ERROR: "python" returned error code 1.
`

0xc0170, 2017-03-20:
cc @theotherjimmy @stevew817

stevew817, 2017-03-20:
@0xc0170 Did you guys break the exporter interface? I'm tracing the arguments I'm getting into the exporter, and am coming up with no files, and only a mbed_config header being passed to the exporter's export() function.

chrissnow, 2017-03-20:
Perhaps related to https://github.com/ARMmbed/mbed-os/issues/3938 ?

stevew817, 2017-03-20:
@chrissnow thanks, that explains a lot.

chrissnow, 2017-03-20:
Indeed the changes in https://github.com/ARMmbed/mbed-os/pull/3924 fix the export for uVision at least,
built successful also.

0xc0170, 2017-03-21:
> Indeed the changes in #3924 fix the export for uVision at least,
built successful also.

Should be integrated today. Thanks for the references here. 

@stevew817 Please just fetch that fix and should fix exporters and allow you to develop, until this fix is integrated to master

theotherjimmy, 2017-04-26:
@chrissnow is this still a problem? if not, could we close this issue?

closed 2017-04-26 by chrissnow