ARMmbed/mbed-os #3964: Export for GNU ARM Eclipse fails with exception

ilg-ul, 2017-03-17:
### Description
- Type: Bug 
- Related issue: `#3561`
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**
not relevant

**mbed-cli version:** 
1.0.0

**meed-os sha:**
f4864dc Merge pull request #3930 from ARMmbed/release-candidate

**DAPLink version:**

**Expected behavior**
`mbed export -m K64F -i gnuarmeclips` to create the new project

**Actual behavior**
the command fails with an exception

**Steps to reproduce**

```
mbed import mbed-os-example-blinky
cd mbed-os-example-blinky/
mbed export -m K64F -i gnuarmeclipse
```

The result is 

```
Scan: .
Traceback (most recent call last):
  File "/Users/ilg/tmp/mb/mbed-os-example-blinky/mbed-os/tools/project.py", line 248, in <module>
    main()
  File "/Users/ilg/tmp/mb/mbed-os-example-blinky/mbed-os/tools/project.py", line 244, in main
    build_profile=profile)
  File "/Users/ilg/tmp/mb/mbed-os-example-blinky/mbed-os/tools/project.py", line 94, in export
    build_profile=build_profile, silent=silent)
  File "/Users/ilg/tmp/mb/mbed-os-example-blinky/mbed-os/tools/project_api.py", line 222, in export_project
    macros=macros)
  File "/Users/ilg/tmp/mb/mbed-os-example-blinky/mbed-os/tools/project_api.py", line 89, in generate_project_files
    exporter.generate()
  File "/Users/ilg/tmp/mb/mbed-os-example-blinky/mbed-os/tools/export/gnuarmeclipse/__init__.py", line 129, in generate
    raise NotSupportedException("No linker script found.")
tools.utils.NotSupportedException: No linker script found.
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /Users/ilg/tmp/mb/mbed-os-example-blinky/mbed-os/tools/project.py -i gnuarmeclipse -m K64F --source ." in "/Users/ilg/tmp/mb/mbed-os-example-blinky"
```

CarloMara, 2017-03-18:
@ilg-ul Shoudn't this be filed under [mbed-cli](https://github.com/ARMmbed/mbed-cli)?

ilg-ul, 2017-03-18:
the exporter sources are part of the mbed-os repo: 

https://github.com/ARMmbed/mbed-os/tree/master/tools/export/gnuarmeclipse

ilg-ul, 2017-03-18:
anyway, it is puzzling that such things happen, with so many and elaborated continuous integration bots running. :-(

CarloMara, 2017-03-18:
I think it's  #3852  @theotherjimmy

theotherjimmy, 2017-03-20:
https://github.com/ARMmbed/mbed-os/pull/3924 is the fix. It's not in yet.

CarloMara, 2017-03-21:
@theotherjimmy do you have a merge ETA?

Carlo

0xc0170, 2017-03-21:
It should happen today

CarloMara, 2017-03-21:
Lovely, thanks

bridadan, 2017-03-23:
Hey y'all, just as a follow up I'd like to mention that these kinds of failures should be caught sooner in the near future. We currently have a few different jobs that we run on PRs depending on what they modify. The most common one we run doesn't include exporters, however I am working to add this in. So hopefully we can avoid these issues in the future :+1:

ilg-ul, 2017-03-23:
any idea when the fix for this bug will be publicly available? I guess we need to wait for the next release. when is it due?

bridadan, 2017-03-23:
5.4.1 was released about a week ago, so I would expect the patch to be showing up publicly within the next two weeks or so. We publish patch releases usually every two weeks (give or take).

CarloMara, 2017-03-23:
Hey @bridadan could you explain me what do you mean with `We currently have a few different jobs that we run on PRs depending on what they modify.`?
I guess it's some kind of automatic testing, am I right?

Cheers, 
Carlo

bridadan, 2017-03-24:
Yup, when you look at open PRs you'll see a bunch of automatic testing that occurs. The results are at the bottom of the PR. We run a number of jobs on them, sometimes not always the same ones. This will help reduce the gap between the jobs that we run, limiting the issues that can slip through in the future.

ilg-ul, 2017-04-26:
with the current version, the export seems functional now.

closed 2017-04-26 by ilg-ul