ARMmbed/mbed-os #3967: Online Compiler does not show Flash/RAM utilization on Nucleo L073RZ & L152RE

helmut64, 2017-03-19:
I believe more targets are affected, my STM32L4 works fine, but the Nucleo L073RZ & L152RE does not show any utilization. See attached picture.

I am using classic mbed #136

Regards Helmut
[image]

0xc0170, 2017-03-20:
cc @theotherjimmy

theotherjimmy, 2017-03-20:
huh. I'll look into this. Thanks for reporting this @helmut64

vargham, 2017-03-21:
Same for STM32F103.

jimeixuehua, 2017-03-23:
Same for STM32F103.

helmut64, 2017-04-08:
Today I figured out that the the TARGET_NUCLEO_L432KC platform shows totally wrong flash sizes (not zero, but wired numbers). It would be great if somebody can address this. The NUCLEO_L476RG target another STM32L4 shows correct sizes.

0xc0170, 2017-04-10:
@theotherjimmy Can you please help?

theotherjimmy, 2017-04-10:
I'm still in the dark on what's causing this. It looks like mbed-cli uses something different here. I'll see what I can do about making them the same.

helmut64, 2017-04-10:
What is strange, the TARGET_NUCLEO_L432KC shows something wired, the NUCLEO_L476RG  (same MCU type) works correct.

The zero Info for the L0 and F1 targets may be something different.

theotherjimmy, 2017-04-10:
Could you expand on "something weird" @helmut64?

helmut64, 2017-04-11:
[image]
[image]

Some project using the STM32 L4 MCU, the NUCLEO-476RG data is correct.

helmut64, 2017-05-01:
Thanks for the fix, I verified it with targets TARGET_NUCLEO_L432KC, TARGET_NUCLEO_L073RZ and TARGET_NUCLEO_F103RB. It works great again.
Regards Helmut

closed 2017-05-01 by helmut64