ARMmbed/mbed-os #3970: Nucleo F303K8 DAC outpud doesn't work

CarloMara, 2017-03-20:
### Testing the NUCLEO F303K8 board from ST, the DAC outputs aren't working

- Type: Bug 
- Priority: Blocker

---------------------------------------------------------------

**Target**
NUCLEO F303K8
All of the test are conducted using mbed online IDE

**Expected behavior**
While testing the analog out Hello World example the DAC output should swing between the maximum value and the minimum while printing on the serial port the DAC value

**Actual behavior**
The output stays fixed at 0V. The readings on the serial port are also stuck at 0V

**Steps to reproduce**
Go to the AnalogOut handbook page, import the `Hello World!` into the web compiler, open the serial port to see the readings.

0xc0170, 2017-03-20:
cc @bcostm @adustm @LMESTM @jeromecoutant

jeromecoutant, 2017-03-20:
Hi
Behavior is OK if you select:
AnalogOut  aout(A3);

Issue occurs when A4 or A5 is selected...
Check is on going

CarloMara, 2017-03-28:
Hi @jeromecoutant ,

Do you have any news on this side?

I have found also this [link](https://translate.google.com/translate?sl=auto&tl=en&js=y&prev=_t&hl=en&ie=UTF-8&u=https%3A%2F%2Fdeveloper.mbed.org%2Fquestions%2F67997%2FNUCLEO-F303K8ADC%2F&edit-text=&act=url), which suggests other problems with the DACs.
Here is the [non translated version](https://developer.mbed.org/questions/67997/NUCLEO-F303K8ADC/).

Thanks again,
Carlo

LMESTM, 2017-06-13:
@CarloMara 

I think I've addressed the issue on DAC in here: #4529 
I suggest that you try out these 3 patches and let me know.

I think the key point in your case is in particular the DAC configuration:
https://github.com/ARMmbed/mbed-os/pull/4529/commits/18d1e8e27897d2d76fdcbcb40029b8370cb3a3d4

CarloMara, 2017-06-13:
I'll try that tomorrow and ping you with the result.

Thanks,
Carlo

LMESTM, 2017-07-03:
@CarloMara @0xc0170  Fix #4529 has now been merged - I propose to close this issue (and re-open in case @CarloMara reports remaining problems)
ST_TO_BE_CLOSED

closed 2017-07-03 by 0xc0170