ARMmbed/mbed-os #3971: Code does not run on UBLOX C027 since 5.2.0

sarahmarshy, 2017-03-20:
### Description
- Type: Bug 
- Priority:Major 

---------------------------------------------------------------
## Bug

**Target**
UBLOX C027

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-os sha:**
e435a07 Merge pull request #3111 from ARMmbed/release-candidate

**DAPLink version:**
```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 12340000fb331bba00000000000000000000000097969902
HIC ID: 97969902
Auto Reset: 0
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0243
Git SHA: b403a07e3696cee1e116d44cbdd64446e056ce38
Local Mods: 0
USB Interfaces: MSD, CDC, HID
Interface CRC: 0x90ebd3e6
Remount count: 32
```

**Steps to reproduce**
Try any program, and it will not run. Though, everything compiles. Debuggers crash, so I guess the device is also crashing. Everything compiles, but It's hard to narrow down the exact commit in this release that broke things, but I've narrowed it down to this release.

@sg- @ansondtx20

BlackstoneEngineering, 2017-03-20:
So 5.2.0 works but everything after it does not, correct?

sarahmarshy, 2017-03-21:
No, 5.2.0 is what broke things.

0xc0170, 2017-03-21:
cc @RobMeades @andreaslarssonublox

RobMeades, 2017-03-22:
Hi there: I think this is probably the `wait_ms(50)` issue in `C027_api.c`.  At least, I've just compiled blinky for headrev mbed-os and UBLOX_C027 and it runs if I make the following change:

In the file `mbed-os\targets\TARGET_NXP\TARGET_LPC176X\TARGET_UBLOX_C027\C027_api.c`, function `c027_init()`, replace the line:

`wait_ms(50);`

...with:

```
uint32_t start = us_ticker_read();
while ((us_ticker_read() - start) < 50000);
```

To have it compile without warnings, `#include "us_ticker_api.h"`.

At some point `wait_ms()` became an RTOS function when it wasn't previously; presumably this happened at mbed 5.2.0.  @hasnainvirk had intended to submit this fix but I can also do so. @0xc0170: really need to get a C027 board into your regression test system.

hasnainvirk, 2017-03-22:
@RobMeades yeah, please go ahead. It would be one less commit from my PR if you go ahead with this :)

andreaslarssonublox, 2017-03-23:
@RobMeades might this PR https://github.com/ARMmbed/mbed-os/issues/3515 be related?

0xc0170, 2017-04-25:
Fixed via https://github.com/ARMmbed/mbed-os/pull/3990

closed 2017-04-25 by 0xc0170