ARMmbed/mbed-os #3977: The BLE UARTService class isn't a Stream

seppestas, 2017-03-21:
Even though [the UARTService.h includes Stream.h](https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_BLE/ble/services/UARTService.h#L25) and [the `UARTService` class implements the `_putc` and `_getc` functions](https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_BLE/ble/services/UARTService.h#L150), the UARTService doesn't actually override the `Stream` class.

This means it's not possible to use handy functions like `printf` on `UARTService` objects.

It seems like [at some point the `UARTService` class](https://developer.mbed.org/teams/Bluetooth-Low-Energy/code/BLE_UARTConsole/file/2130d7f559dc/UARTService.h) did override the `Stream` class, but this functionality was removed when the UARTService was added to the BLE API library.

I think the `UARTService` class should override the `Stream` class or more suitable `read` and `write` methods should be made available on the `UARTService` class. At the very least the comments stating _putc and _getc override Stream::_putc() and Stream::_getc should be removed.

0xc0170, 2017-03-22:
cc @pan-

seppestas, 2017-03-23:
Can I create a PR to make the UARTService a Stream again?

closed 2017-10-27 by ChiefBureaucraticOfficer