ARMmbed/mbed-os #3978: NXP K66F linker files are not up to date

SeppoTakalo, 2017-03-21:
### Description
- Type: Bug
- Priority: Major

I cannot compile Nanostack border router for K66F.
With the same RAM size, K64F works.

```
Link: nanostack-border-router
/Users/septak01/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: ./BUILD/K66F/GCC_ARM/nanostack-border-router.elf section `.bss' will not fit in region `m_data'
/Users/septak01/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: region `m_data' overflowed by 19236 bytes
collect2: error: ld returned 1 exit status
[ERROR] /Users/septak01/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: ./BUILD/K66F/GCC_ARM/nanostack-border-router.elf section `.bss' will not fit in region `m_data'
/Users/septak01/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: region `m_data' overflowed by 19236 bytes
collect2: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /Users/septak01/src/nanostack-border-router/mbed-os/tools/make.py -t GCC_ARM -m K66F --source . --build ./BUILD/K66F/GCC_ARM" in "/Users/septak01/src/nanostack-border-router"
---
```
## Bug

**Target**
K66F

**Toolchain:**
GCC_ARM

0xc0170, 2017-03-21:
Have you compared linker scripts? That they are the same (using same areas for data?). 

I quickly compared, the heap size is different (bigger for K66F by ~40k  that could explain this failure?). BSS for K66F is in m_data section as well (K64F m_data_2).

SeppoTakalo, 2017-03-21:
No, I did not compare.
As a workaround I overwrote my K66F linker file with K64F one. Seemed to work OK and I was able to finnish my work. Seems that these devices have closely same memory layout (but bigger flash).

0xc0170, 2017-03-22:
@mmahadevan108 - please can you have a look at my comment above? And look at the linker script for K66F. Why it has those 2 differences there

TuomoHautamaki, 2017-04-11:
@0xc0170 , is the issue now fixed?

closed 2017-04-11 by 0xc0170

SeppoTakalo, 2017-04-11:
Nanostack border router now compiles with mbed OS 5.4.3