ARMmbed/mbed-os #3983: LocalFileSystem stop working in mbed_lib_rev138

toyowata, 2017-03-22:
### Description
- Type: Bug
- Priority: Blocker
- Reported: https://developer.mbed.org/questions/77498/mbed-local-file-system-not-working/

---------------------------------------------------------------
## Bug

**Target**
LPC1768 (and possibly LPC11U24)

**Toolchain:**
ARM (online IDE)

**Toolchain version:**
5.06

**meed-os sha:**
f4864dc6429e1ff5474111d4e0f6bee36a759b1c

**Expected behavior**
LocalFileSystem API should work.

**Actual behavior**
LocalFileSystem API does not seem to work with mbed_lib_rev138.
If I rollback to mbed_lib_rev137, it works fine.

**Steps to reproduce**
Create new problem as code below with mbed_lib_rev138.

```
#include "mbed.h"

#if !defined(TARGET_LPC1768) && !defined(TARGET_LPC11U24)
#error This target does not support LocalFileSystem.
#endif

LocalFileSystem local("local");               // Create the local filesystem under the name "local"
 
int main() {
    FILE *fp = fopen("/local/out.txt", "w");  // Open "out.txt" on the local file system for writing
    fprintf(fp, "Hello World!");
    fclose(fp);
}
```

0xc0170, 2017-03-22:
@toyowata Any pointers ? I checked the history, local filesystem and lpc176x have not been changed for some time.

toyowata, 2017-03-22:
@0xc0170 Not sure the trigger of this problem, but I got one warning message with mbed_lib_rev138 build which was not produced by previous library.

```
Warning: Function "mbed::FileSystemLike::FileSystemLike(const char *)" (declared at <a href="#" onmousedown="mbed_doc_goto('/LocalFileSystem_HelloWorld//extras/mbed_093f2bd7b9eb/drivers/FileSystemLike.h', '47'); return false;">/extras/mbed_093f2bd7b9eb/drivers/FileSystemLike.h:47</a>) was declared "deprecated" in "extras/mbed_093f2bd7b9eb/drivers/LocalFileSystem.h", Line: 99, Col: 53
```

toyowata, 2017-03-27:
@0xc0170 I saw that the FileLike class had been modified between rev137 and rev138.  It does not use FileHandle class any more, but not sure this is a cause of this problem.

cc @geky

closed 2017-06-04 by sg-