ARMmbed/mbed-os #3984: MCU_NRF52: CONFIG_GPIO_AS_PINRESET should be a config option not platform macro

jaustin, 2017-03-22:
in targets.json, for MCU_NRF52 there are a number of useful config options.

Unfortunately, the configuration of the NRESET pin as GPIO or not is not one of those options.

It'd be nice if we could have a '1'/'0' config option. I think this would mean changing mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/device/system_nrf52.c to check for the value rather than just for the definition, but not sure how config handles things that can be defined or not defined and default values, etc.

0xc0170, 2017-03-22:
cc @anangl @nvlsianpu @pan-

nvlsianpu, 2017-03-22:
Thanks for input. We will consider this suggestion as one more gpio is usefull for the App.

closed 2017-10-27 by ChiefBureaucraticOfficer