ARMmbed/mbed-os #3985: config: macros_remove doesn't remove CONFIG_GPIO_AS_PINRESET macro for NRF52_DK

jaustin, 2017-03-22:
Hiya,

It seems that using target.macros_remove in app_config.json does not stop the CONFIG_GPIO_AS_PINRESET being included on the NRF52_DK target:

`$cat mbed_app.json`
```
{
     "target_overrides": {
        "*": {
		"target.macros_remove": "CONFIG_GPIO_AS_PINRESET"
        }
     }
}
```
`$mbed compile -vv`
```
[mbed] Working path "[...]/mbed-os-example-blinky" (program)
[mbed] Exec "python -u [...]/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m nrf52_dk --source . --build ./BUILD/nrf52_dk/GCC_ARM -v" in [...]/mbed-os-example-blinky
Building project mbed-os-example-blinky (NRF52_DK, GCC_ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
[DEBUG] Macros: -DDEVICE_ERROR_PATTERN=1 -DTARGET_MCU_NRF52832 -DNRF52_PAN_51 -D__MBED__=1 -DNRF52_PAN_53 -D__FPU_PRESENT=1 -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DBOARD_PCA10040 -DNRF52_PAN_12 -DTOOLCHAIN_GCC -DNRF52_PAN_58 -DDEVICE_SPISLAVE=1 -DNRF52_PAN_55 -DNRF52_PAN_54 -DTARGET_CORTEX_M -DNRF52_PAN_30 -DTARGET_NRF52_DK -DNRF52_PAN_36 -DTARGET_LIKE_CORTEX_M4 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NRF5 -DFEATURE_BLE=1 -DTARGET_MCU_NRF52 -DNRF52_PAN_31 -DDEVICE_SERIAL=1 -DDEVICE_SPI_ASYNCH=1 -DNRF52 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DTARGET_NRF52832 -D__CORTEX_M4 -DTARGET_LIKE_MBED -DNRF52_PAN_63 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_NORDIC -DDEVICE_SERIAL_FC=1 -DBLE_STACK_SUPPORT_REQD -DNRF52_PAN_15 -DCONFIG_GPIO_AS_PINRESET -DMBED_BUILD_TIMESTAMP=1490149955.71 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DSWI_DISABLE0 -DNRF52_PAN_20 -DDEVICE_SPI=1 -DDEVICE_INTERRUPTIN=1 -DNRF52_PAN_64 -DSOFTDEVICE_PRESENT -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DNRF52_PAN_62 -DS132 -DDEVICE_I2C_ASYNCH=1 -DARM_MATH_CM4
```

This appears to be because of the way target_overrides is called. I've proposed a fix that works for me here jaustin/mbed-os-1@02cf914eb007a0dae01cc5ed6d0c27be97470edc (https://github.com/jaustin/mbed-os-1/commit/02cf914eb007a0dae01cc5ed6d0c27be97470edc)

...but this is a fairly naive fix so happy to be correct @theotherjimmy @screamerbg - perhaps I'm using the app_config.json incorrectly?

theotherjimmy, 2017-03-22:
Oh, @jaustin You should be using a list of macros to remove. 
so 
```json
{
     "target_overrides": {
        "*": {
		"target.macros_remove": ["CONFIG_GPIO_AS_PINRESET"]
        }
     }
}
```

but we can look into up-converting the singleton to a list with one element if that would be reasonable.

jaustin, 2017-03-22:
Sigh. That makes sense. 

I think we should at least throw an error in the case of no list. Converting the single string feels like spooky action at a distance but if we can't error then it's probably preferable

theotherjimmy, 2017-03-22:
We can do either. Which would you prefer?

jaustin, 2017-03-22:
I think it's best to throw an error with a hint as to what the correct
syntax is -keep the logic simple.

Actually more errors in general would be nice. I found lots of ways to feed
invalid data to that config without anything complaining, even missing
commas are silently accepted iirc

J

theotherjimmy, 2017-03-22:
Listing them here would help me track this down and do better validation in the tools. I'll make the particular corner case you found an error.

closed 2017-03-29 by sg-