ARMmbed/mbed-os #3986: Cmsis 5:references to rtx in rtos cpp wrappers

YarivCol, 2017-03-22:
### Description
- Type: Enhancement
- Related issue: `#rtos`
----------------------------------------------------------------
## Enhancement

In the cpp wrappers of rtos, there are references to rtx os and not only cmsis_os2.h.

If the warppers will reference only cmsis_os2.h the kernel can be replaced with ease (only replace rtx folder with the other os).

0xc0170, 2017-03-22:
It would be helpful to pinpoint some headers that are pulled from rtx, and a reason why (if there are any rtx types used or any specific rtx things). I found in most headers inside rtos folder - inclusion of ``#include "rtx_lib.h"``.
> If the warppers will reference only cmsis_os2.h the kernel can be replaced with ease (only replace rtx folder with the other os).

It would be nice to have

@bulislaw

bulislaw, 2017-03-22:
We are including `#include "rtx_lib.h"` so we can declare memory for the RTOS primitives using internal RTX objects like https://github.com/ARMmbed/mbed-os/blob/feature_cmsis5/rtos/Mutex.h#L62
I find it cleaner and less of a burden to maintain and/or update than something like `int32_t _mutex_data[4];` at the same time it's pretty much necessary as we don't generally support RTX memory models, also it's clearer (and easier to track) to commit memory from the stack (or global) when object is declared rather than use some pools/carveouts with lifetime of the program.

YarivCol, 2017-03-22:
I understand and agree.
But maybe include "rtx_lib.h" in "cmsis_os2.h",
and every kernel will have similar defines.
So there will not be references to rtx other than cmsis_os2.h.

bulislaw, 2017-03-23:
`rtx_lib.h` is an internal RTX file afaik, and `cmsis_os2.h` is a public interface file, I don't think Keil guys would be willing to make that change. In any case users should be isolated from that problem, as it's internal mbed OS issue in case of an upgrade, not to mention that names/size should be different for other rtoses. What problem are you trying to solve?

YarivCol, 2017-03-23:
ן want to switch to other rtos kernel(not rtx) in my project.
I can solve it by adding compile condtion, but i thought about more generic solution to mbed itself.

bulislaw, 2017-03-23:
Are the function/object names compatible?

YarivCol, 2017-03-23:
Yes, it is compatible.
 I implemented cmsis_os2.h using that kernel api.
 I declared same the names, you used to allocate memory for control blocks.

bulislaw, 2017-03-23:
Right, if you declared the same names and implemented cmsis_os2.h can't you also create/update rtx_lib.h for your purposes? My point is mbed OS uses Keil RTX, RTOSes are different anyway and have different API so 'drop in' replacement is not something that can be usually done without any modification or shim adaptation layers. I don't think your use case is something that most users would care and if they happen to have specific requirement for the RTOS used, they will need to use the API will need adaptation anyway.

0xc0170, 2017-03-23:
> Right, if you declared the same names and implemented cmsis_os2.h can't you also create/update rtx_lib.h for your purposes? My point is mbed OS uses Keil RTX, RTOSes are different anyway and have [...]

That's what CMSIS RTOS should do: From the CMSIS RTOS 2 docs: ``The CMSIS-RTOS API Version 2 (CMSIS-RTOS2) is a generic RTOS interface for ARM® Cortex®-M processor-based devices.`` Therefore I wonder why there's no cmsis objects like you pinpointed earlier (``os_mutex_t``).  We used to use CMSIS-RTOS API as much as we could. If there are reasons for using directly RTX internals, we should document them and share with CMSIS team (where CMSIS RTOS fail or we fail?).

I can't find the examples for CMSIS RTOS 2 as they have for v1 : https://github.com/ARM-software/CMSIS_5/blob/4da2d09bb336e1c08ad54fa9988658eccaa8d169/CMSIS/RTOS/Template/CPP/Mutex.h#L37. I cant recall why that mutex data are protected with that macro:

```
#ifdef CMSIS_OS_RTX
    int32_t _mutex_data[3];
#endif
```

closed 2017-10-27 by ChiefBureaucraticOfficer