ARMmbed/mbed-os #3989: When using callback(): error: too many arguments to function

janjongboom, 2017-03-22:
### Description
- Type: Bug
- Priority: Minor

## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3

**meed-os sha:**
5.3.6

I'm having this file: [accelerometer.h](https://github.com/ARMmbed/workshop-mbed-connect-asia-lights/blob/master/accelerometer.h) which I'm updating to mbed OS 5.3.6. When I write:

```cpp
        eventThread.start(callback(this, &Accelerometer::start_thread));
```

I get a compile error: `too many arguments to function`.

When I do:

```cpp
        Callback<void()> whatever(this, &Accelerometer::start_thread);
        eventThread.start(whatever);
```

It does compile.

@geky

geky, 2017-03-22:
Hi @janjongboom, it's unfortunately picking up the member variable with the same name `callback`:
https://github.com/ARMmbed/workshop-mbed-connect-asia-lights/blob/master/accelerometer.h#L50

Using the full namespace'd name avoids this issue:
``` cpp
eventThread.start(mbed::callback(this, &Accelerometer::start_thread));
```

sg-, 2017-03-24:
or renaming ```void (*callback)(void);``` to ```void (*cb)(void);``` for example in https://github.com/ARMmbed/workshop-mbed-connect-asia-lights/blob/master/accelerometer.h#L50

closed 2017-03-24 by sg-

janjongboom, 2017-03-24:
Ah, good catch. Thanks.