ARMmbed/mbed-os #3999: AnalogOut does not work for K64F

mray19027, 2017-03-22:
### Description
- Type: Bug
- Priority: Blocker

AnalogOut does not work for the K64F from mbed-os version 5.2.3 to 5.4.1.
mbed-os version 5.2.2 has a working AnalogOut api.

AnalogOut constructor works, and no errors occur during runtime. But regardless of what value gets set to the AnalogOut pin, the value floats at 0.03V.

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**
gcc version 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

**mbed-cli version:** 
1.0.0

**meed-os sha:**
cc58a7fb0

mray19027, 2017-03-22:
@BlackstoneEngineering @senthilr 

@sg- 
@MarceloSalazar Why didn't OOB catch this? Do we not have any OOB tests that utilize analog out?
@screamerbg We should incorporate the CI test shield into our OOB testing. The CI test shield discovered this bug.

MarceloSalazar, 2017-03-23:
We don't have example apps in OOB to catch these issues.
IMO, testing the CI shield is not something that we should be doing as part of OOB (it's not in the scope and goals of OOB).
The CI shield could be added to the the CI automated testing infrastructure.

0xc0170, 2017-03-23:
>  The CI shield could be added to the the CI automated testing infrastructure.

+1

cc @mmahadevan108

mmahadevan108, 2017-03-24:
@0xc0170 Is this a bug related to the NXP port? Description states this was working before.

mray19027, 2017-03-30:
@mmahadevan108 I believe so.
Other platforms do not have this issue

BlackstoneEngineering, 2017-04-04:
@MarceloSalazar In the medium to long term we should have the shield on the CI Infrstructure. In the short term we should do this as part of OOB on the relevant platforms as we do not yet have it integrated onto the CI farm.

deepikabhavnani, 2017-07-28:
@mmahadevan108 - DAC output is not enabled in source code. If I un-comment the below line in `DAC_Init()` function, DAC works properly.

`/* DAC_Enable(base, true); */
    /* Tip: The DAC output can be enabled till then after user sets their own available data in application. */`

Since the comment says it should be enabled when data is set, can you please fix it at write place and enable DAC.

mmahadevan108, 2017-08-01:
@deepikabhavnani Thank you. I have submitted a PR to fix this issue.

closed 2017-11-29 by deepikabhavnani