ARMmbed/mbed-os #4001: How to set MBED_CONF_NSAPI_PRESENT to 0

TsungtaWu, 2017-03-23:
### Description
- Type: Question
- Priority: Blocker

## Question

For DELTA_DFCM_NNN50 platform, it comes with EthernetInterface lib. It seems there are few conflicts with features\netsocket when compiled. I found one solution is to set MBED_CONF_NSAPI_PRESENT to 0 in mbed.h to avoid coflicts since "netsocket/nsapi.h" would not included. What is the propoer method to this config for specifc target?

0xc0170, 2017-03-23:
cc @geky

geky, 2017-03-23:
I would create a .mbedignore file that removes the nsapi from compilation:

In .mbedignore:
```
mbed-os/features/netsocket/
```

Removing the nsapi from compilation will as a side effect set MBED_CONF_NSAPI_PRESENT to 0. If you want to keep both nsapi and the EthernetInterface.lib you may run into other issues as I believe they both use lwip with static declarations.

sg-, 2017-03-24:
I think EthernetInterface.lib needs to be updated to use NSAPI instead. .mbedignore is not the proper solution.

EthernetInterface.lib is for mbed OS 2, not mbed OS 5. Please don't confuse these or make temporary solutions.

TsungtaWu, 2017-03-24:
@sg- The EthernetInterface.lib that DELTA_DFCM_NNN50 used is not lwip based which is quite different with the one used by K64F and LPC1768. In the other words, the EthernetInterface.lib for DELTA_DFCM_NNN50 is highly customised, we only make it to follow the API syntax defined by mbed easier for our platform user to pick up the WIFI function. We understand that mbed officail EthernetInterface.lib is for mbed OS 2, not mbed OS 5. In our case, since the EthernetInterface.lib is proprietary for DELTA_DFCM_NNN50, does it allow to be used with mbed OS 5?

sg-, 2017-03-24:
All mbed OS 5 networking interfaces should be done with NSAPI regardless if in tree based on lwIP or Nanostack or out of tree based on a proprietary stack.

closed 2017-10-27 by ChiefBureaucraticOfficer