ARMmbed/mbed-os #4007: EFM32PG thread never wakes

chrissnow, 2017-03-23:
- Type: Bug 
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
efm32pg_stk3401

**Toolchain:**
ARM

**Toolchain version:**

**mbed-cli version:** 
1.0.0

If all threads are idle and the idle thread gets called which calls sleep() threads never wake up.
the below code demonstrates the problem.

It needs to be compiled in release and after each program the power to the target cycled, reset is not enough as it seems to prevent sleep.

if the while(1) is called which prevents an idle it works as expected.

Expected results.
LED0 toggles every 10 seconds or when BTN1 is pressed.
LED1 toggles when BTN1 is pressed

Actual results AFTER POWER cycle.
LED0 and LED1 toggles when BTN1 is pressed

Actual results AFTER PROGRAM.
LED0 toggles every 10 seconds or when BTN1 is pressed.
LED1 toggles when BTN1 is pressed

```c++
#include "mbed.h"

DigitalOut led1(LED1);
DigitalOut led0(LED0);
InterruptIn button(BTN1);

Thread testThread;

void testTask(){
	while(true)
	{
		Thread::signal_wait(0x01,10000);
		led0=!led0;
	}
}

void Button() {
	testThread.signal_set(0x01);
	led1=!led1;
}

int main(void) {
	button.fall(&Button); // attach the address of the flip function to the rising edge
	testThread.start(callback(testTask));
	
//no sleeping allowed if the below line is called
	//while (1);

}
```

0xc0170, 2017-03-23:
cc @stevew817

stevew817, 2017-03-23:
@0xc0170 This looks to me like a general problem for everyone using RTOS on mbedOS 5. The RTOS scheduler is running on the systick timer, which gets clocked by the core clock. When you call sleep, that clock gets turned off, so the timer doesn't have a clock anymore, so it won't fire an interrupt into the scheduler anymore.

Quick fix is probably to not allow sleeping when using an RTOS? This is why we need tickless support.

chrissnow, 2017-04-24:
@stevew817 can you think of any other way around this? perhaps use a timer to periodically wake the device?

It's a battery powered device and makes use of threads, so disabling sleep isn't really an option.
timing isn't critical but the threads must at least function!

stevew817, 2017-04-25:
@chrissnow Can you try using a LowPowerTicker instead of calling wait() on a thread? That will actually set up a periodic interrupt for you, and should result in the expected behaviour.

stevew817, 2017-12-13:
@0xc0170 This should be resolved by tickless and the move to the mbed sleep manager.
@chrissnow Can you verify this is no longer an issue and close the ticket?

stevew817, 2018-03-08:
@0xc0170 or @cmonr Please close this issue since it should have been fixed and the ticket has been inactive for a very long time.

closed 2018-03-08 by 0xc0170