ARMmbed/mbed-os #4009: Communication speed is very slow due to lock / unlock in 1 byte unit.

TomoYamanaka, 2017-03-23:
### Description
- Type: Question
- Priority: Major

---------------------------------------------------------------
## Question

Since mbed OS 5, Communication speed is very slow due to lock / unlock in 1 byte unit.
For example SPI driver:
Because of `lock()` and `unlock()` which are called in the following two functions, 
`lock()` and `unlock()` with mutex function are executed twice every 1 byte transfer. 

<SPI.cpp>
```
void SPI::aquire() {
    lock();
     if (_owner != this) {
        spi_format(&_spi, _bits, _mode, 0);
        spi_frequency(&_spi, _hz);
        _owner = this;
    }
    unlock();
}
```
```
int SPI::write(int value) {
    lock();
    aquire();
    int ret = spi_master_write(&_spi, value);
    unlock();
    return ret;
}
```
When accessing SD by using SPI driver, the communication speed is very slow compared with mbed classic speed. Because it takes time for SD access, music transfer and image transfer may fail.
In mbed classic, communication speed is fast because of `lock()` / `unlock()` with empty function.

In SPI communication, I guess that communication of only 1 byte is rare.
Therefore, it's necessary to lock at application layer so that it will not be interrupted in the middle of communication.
In this case, it's not necessary to lock / unlock in 1 byte unit.
The following `sd-driver` lock the part that not be interrupted in the middle of communication at application layer.
https://github.com/ARMmbed/sd-driver

```
int SDBlockDevice::_cmd(int cmd, int arg) {
    _spi.lock();
    _cs = 0;

    // send a command
    _spi.write(0x40 | cmd);
    _spi.write(arg >> 24);
    _spi.write(arg >> 16);
    _spi.write(arg >> 8);
    _spi.write(arg >> 0);
    _spi.write(0x95);

    // wait for the repsonse (response[7] == 0)
    for (int i = 0; i < SD_COMMAND_TIMEOUT; i++) {
        int response = _spi.write(0xFF);
        if (!(response & 0x80)) {
            _cs = 1;
            _spi.write(0xFF);
            _spi.unlock();
            return response;
        }
    }

    _cs = 1;
    _spi.write(0xFF);
    _spi.unlock();
    return -1; // timeout
}
```

simonqhughes, 2017-03-30:
Yes, you're correct about the double locking and SDBlockDevice should be improved. I did the following test to further understand the impact:
- implement class SDBLockDeviceSPI: public SPI with overrides for lock() and unlock() with null implementations. 
- Changed the SDBlockDevice::_spi type to be SDBLockDeviceSPI
- measured the fopen() test case 16 performance.

The above takes out the spi locking per SPI::write() and per SPI:aquire() call for SDBlockDevice operations. 

Before the changes the time to run the fopen test case 16 was ~82s. After the changes the test time was ~49s (approx. 40% improvement).

Thanks

TomoYamanaka, 2017-03-31:
@simonqhughes 
Thank you for your reply.
<br>

We are considering about solution for this issue. Please refer to the following it.
https://developer.mbed.org/users/dkato/code/sd-driver-hs/file/b22a1df967cb/features/filesystem/sd/SDBlockDevice.h

This file is based on the below. There is a difference between after line 130 of this and after line 129 of ours.
https://github.com/ARMmbed/sd-driver/blob/master/features/filesystem/sd/SDBlockDevice.h

The same as your suggestion, it overrides public SPI. In case of ours, it overrides `write() `and `aquire()`. 
Therefore, 
`lock()` and `unlock()` are called **just once**, communication speed is **not slow**. In case of the code in my first comments, `_spi.lock()` and `_spi.unlock()` are called, however lock / unlock in 1 byte unit are not called in SPI driver.

Could you give us your opinion about the above ?
<br>

In addition,
We are changing the access restriction from `privste:` to `protected:` in ours.
https://developer.mbed.org/users/dkato/code/sd-driver-hs/file/b22a1df967cb/features/filesystem/sd/SDBlockDevice.h (line 129)
https://github.com/ARMmbed/sd-driver/blob/master/features/filesystem/sd/SDBlockDevice.h (line 130)

This is reason that we would like to change `_transfer_sck` **from outside** by inheriting class. We would like to change the clock to about 20 MHz and access it fast.

Could you give us your opinion about the above too?
<br>

TomoYamanaka, 2017-04-17:
How is this going ?

TomoYamanaka, 2017-04-21:
Related PR is here.
https://github.com/ARMmbed/mbed-os/pull/4207

closed 2017-07-20 by TomoYamanaka