ARMmbed/mbed-os #4018: LPC1768 DHCP

chrissnow, 2017-03-24:
- Type: Bug 
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
LPC1768

**Toolchain:**
ARM

**Toolchain version:**

**mbed-cli version:** 
1.0.0

Since mbed-os 5.4.0 DHCP appears to be broken.
I can see DHCP request and offers with wireshark but no IP is assigned to the interface
Statically assigned IP works fine.

toyowata, 2017-03-24:
Related this issue:
https://github.com/ARMmbed/mbed-os-example-sockets/issues/17

geky, 2017-03-27:
I'm having a difficult time reproducing this issue. The LPC1768 and ARCH_PRO both seem to work on the two networks I have available.

Could you attach the wireshark dump of the DHCP request/offer? I can probably fake the traffic to reproduce it here.

Also what does your ethernet hardware setup look like? If you are using an [Arch Pro](https://developer.mbed.org/platforms/Seeeduino-Arch-Pro/), did you compile with `mbed compile -m ARCH_PRO`?

chrissnow, 2017-03-27:
I will get you a wireshark capture tomorrow, I may try a Ublox C027 I have at home tonight which is basically the same hardware.
I'm using the EA LPC1769 board and a magjack.
https://www.embeddedartists.com/products/lpcxpresso/lpc1769_cmsis_xpr.php
mbed compile -t arm -m lpc1768

since going to static IP I haven't had any issues so it's unlikely to be hardware.

chrissnow, 2017-03-28:
@geky here is a capture
[dhcp.zip](https://github.com/ARMmbed/mbed-os/files/875484/dhcp.zip)

geky, 2017-03-28:
Thanks! that will help a bunch.

Starting to dig my teeth in, it looks like the difference between our DHCP servers is that yours is _responding_ with broadcast packets. This is likely an issue with a recent change to add broadcast support (https://github.com/ARMmbed/mbed-os/pull/3482?)

geky, 2017-03-28:
@chrissnow, I think I found the issue in https://github.com/ARMmbed/mbed-os/pull/4060, let us know if there is still an issue.

closed 2017-10-11 by chrissnow