ARMmbed/mbed-os #4023: filesystem: block device api is too specific

YarivCol, 2017-03-24:
### Description
- Type: Enhancement

----------------------------------------------------------------
## Enhancement

The api of block device have program and erase function instead of simply write function:
`virtual int program(const void *buffer, bd_addr_t addr, bd_size_t size) = 0;`
`virtual int erase(bd_addr_t addr, bd_size_t size) = 0;`

instead of:
`virtual int write(const void *buffer, bd_addr_t addr, bd_size_t size) = 0;`

 also i think only two size function will be sufficient:

`virtual bd_size_t size() const = 0;`
`virtual bd_size_t get_block_size() const = 0;`

the main motivation of this enhancement is to change the api to look more like proven block device api like the linux kernel one.
Also for encapsulate reasons: files system needs only to write to device without knowing how the device itself works (erase, program), even more in current fat wrapper in each write erase is called causing fast wear in flash block devices (fat file system is not journaling file system so it will try to write to same blocks causing them to wear) and slow write in nor flash devices( erase in nor flash takes time).
if block device will have only write , flash block device can implement [FTL layer](https://flashdba.com/2014/09/17/understanding-flash-the-flash-translation-layer/) and prevent fast wearing of device. 

also i am in favor of creating of FTL block device and flash device(with the same api as current  block device api), the FTL block device will use flash device.

*i am working on this patch.

0xc0170, 2017-03-27:
cc @geky @simonqhughes @c1728p9

c1728p9, 2017-03-27:
cc @theotherjimmy

geky, 2017-03-27:
Well, we do have a need for the current api to describe flash devices (large erase sizes makes combining program+erase problematic at the lower layers). The block device api can also describe generalized block devices by having `get_program_size` return `get_erase_size` and making `erase` a noop (perhaps this should be default). The [sd-driver](https://github.com/ARMmbed/sd-driver/tree/master/features/filesystem/sd) is a good example of this.

Since the `erase` function can be a noop, the current api can abstract both generalized block devices and flash devices. Adding separate classes for BlockDevice and FlashDevice creates a distinction that I'm not really sure is necessary.

Mounting a FAT filesystem to raw flash may not be the most efficient in terms of wear and runtime, but it is perfectly reasonable for a read-heavy application when code-size is the main concern (config information on an embedded system for example).

It's also not unreasonable for a user to mount a flash-filesystem to a general block device such as an SD card. This is a great method for debugging a filesystem, and may provide other features such as fault-tolerance.

We would also need to duplicate the generalized operations such as device [chaining](https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/bd/ChainingBlockDevice.h) and [slicing](https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/bd/SlicingBlockDevice.h).

If we need to provide both a `BlockFlashDevice` and `FlashBlockDevice` I start wondering if the abstraction does us more harm than good.

---

> also i am in favor of creating of FTL block device and flash device(with the same api as current block device api), the FTL block device will use flash device.

I'd be very exciting to see an FTL in mbed OS! I'm looking forward to what you come up with.

YarivCol, 2017-03-27:
my problem is with the names of the functions, in storage (to what i know) erase means change bit to 1 and program means change to 0.
But in SD for example the function program and erase do something else, so it a bit confusing and misleading.

i still think blockdevice should have write and read functions it will allow flexibility without the need to go to the code to know what the function really do.

geky, 2017-03-27:
Do you need to know the state of the storage between erase and program? If you don't, both the SD card (where is erase is noop) and flash (where is erase sets bits to 1) will behave the same.

If you do rely on the "nanding" nature of flash, we were considering adding a `get_erase_value` or similar to allow you to check for the state of an erased block.

We did look into adding a `write` function that just calls erase+program ([prototype](https://github.com/ARMmbed/mbed-os/blob/5a977eff301bfc588a5ae27e202e6cea6e866714/features/filesystem/bd/BlockDevice.cpp#L20-L33)), but having both `write` and `program` caused confusion for users so we decided to drop it. We could add the `write` function back in if it's wanted.

YarivCol, 2017-03-27:
my FTL will need some way to ask flash device for erase block state, my interface for flash has one.

for the second part, i thought of replacing erase and program with write (erase and program  will be handled in the implementation if needed).

closed 2018-01-31 by cmonr