ARMmbed/mbed-os #4024: feature_cmsis5 and osThreadNew problem

Wazalogic, 2017-03-24:
Hello. I am using the "feature_cmsis5" branch because it has improved thread support. But I am having trouble starting threads.

The following example suppose to spawns runMethod(). For some reason, the osThreadNew() method fails and returns NULL. I have tried two different forms (example 1 and example 2) without success. See code below.

```
#include <cmsis_os2.h>

#define example 1

void runMethod(void *parameter) {
      printf("runMethod\r\n");
}

int main() {

#if example == 1
    os_thread_t cb_thread;
    memset(&cb_thread, 0, sizeof(cb_thread));
    osThreadAttr_t thread_attr;
    memset(&thread_attr, 0, sizeof(thread_attr));
    thread_attr.name = "thread";
    thread_attr.stack_mem = new uint8_t[100];
    thread_attr.stack_size = 100;
    thread_attr.priority = osPriorityNormal;
    thread_attr.cb_mem = &cb_thread;
    thread_attr.cb_size = sizeof(os_thread_t);
    osThreadId_t id = osThreadNew((osThreadFunc_t)runMethod, NULL, &thread_attr);
#endif

#if example == 2
    osThreadId_t id = osThreadNew((osThreadFunc_t)runMethod, NULL, NULL);
#endif

    if (id == NULL) {
        printf("Thread not created!\r\n");
    }
   return 0;
}
```

The result is the output "Thread not created!". Any suggestion what may go wrong?

I use a FRDM-K64F and ARM_GCC 6.3.1.

0xc0170, 2017-03-27:
cc @bulislaw

bulislaw, 2017-03-27:
RTX has some assumptions about memory alignment and size, I think you are hitting this one: 
https://github.com/ARMmbed/mbed-os/blob/feature_cmsis5/rtos/rtx2/TARGET_CORTEX_M/rtx_thread.c#L645

Wazalogic, 2017-03-27:
Thanks, 128 seems to work. I will test more tomorrow.
I would have expected that "#define example 2" would work. It's a Keil example.

Wazalogic, 2017-03-28:
Why isn't it allowed to use osThreadNew(), osThreadSuspend() and osThreadResume() in critical regions? See rtx_thread.c lines 1505, 1589 and 1599..

bulislaw, 2017-03-28:
You'll need to ask RTX guys: https://github.com/ARM-software/CMSIS_5

Wazalogic, 2017-03-28:
ah ok. I will ask them, thanks!

bulislaw, 2017-04-05:
Closing it for now, please reopen if you want to follow up.

closed 2017-04-05 by bulislaw