ARMmbed/mbed-os #4026: NRF51822 timer has a race condition

Sissors, 2017-03-24:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NRF51822 / maybe also other NRF but that code is even more confusing

**Expected behavior**
Time increases monotonically until it has 32-bit overflow

**Actual behavior**
Time does not increase monotonically. So either space-time continuum is broken, or the timer code.

**Problem cause**
Now I don't want to tell I told you so, but I did tell you so. Long ago when this version of the timer code was introduced ;).

The easy race condition is the case where interrupts are disabled. Then have a look at this code:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/us_ticker.c#L157

The timer is at 0xFFFFFFF, so no overflow happened. It continues with the code, but before it can read the COUNTER register, an overflow happened. This overflow is not handled, so it will return a time roughly 5 minutes in the past. Sometimes it won't be an issue, but other times it will be disastrous to the program. 

If the interrupt is enabled there is still a race condition between when the COUNTER and overflow variable are read, and when the overflow variable is incremented by the interrupt (based on my experimental results it should first read the overflow variable, and then it reads the COUNTER, so if an overflow happens between those events, the COUNTER is reset, but the overflow variable is not incremented). To verify this 100% you would need to check the assembly code, but this is what my debug code seems to indicate.

**Steps to reproduce**
We cannot write the COUNTER to a specific value. So to test if a racecondition occurs we would need to wait very long. Luckily someone at Nordic realized this, and they added the handy task: TASKS_TRIGOVRFLW. Which forces the timer to 0xFFFFF0. So we can put the clock forward and keep trying until we get an error:

```
#include "mbed.h"

int32_t prev;
int32_t current;

int main() {
    prev = 0;
    us_ticker_read();   //Dummy read to enable the timer
    __disable_irq();
    while(1) {
        
        printf("Time = %u\r\n", us_ticker_read());
        NRF_RTC1->TASKS_TRIGOVRFLW = 1; 

        for (int i = 0; i<10000; i++) {
            current = us_ticker_read();
            if (current - prev < 0) {
                printf("\n\n\rError: New read = %u, error read = %u, previous read = %u\r\n", us_ticker_read(), current, prev);
                while(1);
            }
            prev = current;
        }
    }
}
```
With the TRIGOVRFLW task the timer is set close to the overflow point. Then in a loop as quickly as possible we read the current time. The newly read time should always be equal to or larger than the previous time, so if that is not the case, something went wrong. (32-bit overflows are handled properly by this). If something went wrong, we print the read value, the previous value, and also a new read from the timer. If everything went fine we just trigger the task against and try another time. Since the 32kHz clock and system clock are uncorrelated, in the end we should see the race condition.

Now I am going to add a disclaimer: I do not know if the ones described above are the only issues, since I am surprised at how quickly it goes wrong, often enough at the first attempt already. Still whatever the reason, it goes wrong. An example output of the program:

```
Error: New read = 3584000000, error read = 3072000000, previous read = 3583999969
Time = 30
Time = 512065460
Time = 1024065460
Time = 1536065429
Time = 2048065429

Error: New read = 2560000000, error read = 2048000000, previous read = 2559999969
```

So it read 2559999969. Which should overflow to 2560000000, which is what happens the second time it is read after the overflow. But the first time it returns the situation where COUNTER is zero, but the overflow count variable not yet updated: 2048000000

0xc0170, 2017-03-27:
cc @pan- @nvlsianpu @anangl

pan-, 2017-03-27:
@Sissors Many thanks for the detailed bug report.

> Now I am going to add a disclaimer: I do not know if the ones described above are the only issues, since I am surprised at how quickly it goes wrong, often enough at the first attempt already. 

There is other issues in ticker code (like this [one](https://github.com/ARMmbed/mbed-os/blob/master/hal/mbed_ticker_api.c#L37)) but the read function is directly mapped to the one in the interface (see [here](https://github.com/ARMmbed/mbed-os/blob/master/hal/mbed_ticker_api.c#L119)).

> NRF51822 / maybe also other NRF but that code is even more confusing

Unfortunately, NRF51822 target has received less attention than the NRF5 based targets (NRF51_DK, NRF51_DONGLE and all NRF52 based targets) but both of the targets are subject to the bug reported here. 

> If the interrupt is enabled there is still a race condition between when the COUNTER and overflow variable are read, and when the overflow variable is incremented by the interrupt (based on my [...]

Even if the overflow interrupt is enabled, an overflow can happen during handling of the `CC` interrupt. In my opinion, the solution (overflow interrupt enabled or not) would be something like: 

```c
static inline uint64_t rtc1_getCounter64(void)
{
    uint32_t counter; 
    while (true)  {
        counter = NRF_RTC1->COUNTER;

        if (NRF_RTC1->EVENTS_OVRFLW) {
            overflowCount++;
            NRF_RTC1->EVENTS_OVRFLW = 0;
            NRF_RTC1->EVTENCLR      = RTC_EVTEN_OVRFLW_Msk;
        } else { 
            break;
        }
    }
    return ((uint64_t)overflowCount << 24) | counter;
}
```

The issue is gone with this addition.

@nvlsianpu It would be interesting to complement https://github.com/ARMmbed/mbed-os/pull/4016 with content generated in this issue.

Sissors, 2017-03-27:
@pan- , I think that code is indeed the proper way to handle it without interrupts. But would it be risk free if interrupts are enabled? If the interrupt happens between reading the COUNTER and returning the value, it could still return a wrong timestamp, right?

Good to see the fast reaction to the issue. In general synchronizing a hardware timer with a software part is always tricky business :).

pan-, 2017-03-27:
@Sissors You're right, `rtc1_getCounter64` is not interrupt safe; indeed no operation accessing more than a single register to compute a result is safe. It is possible to protect the transaction by a pair of `__disable_irq` and `__enable_irq`:

```c++
static inline uint64_t rtc1_getCounter64(void)
{
    uint32_t counter; 
    while (true)  {
        __disable_irq();
        counter = NRF_RTC1->COUNTER;

        if (NRF_RTC1->EVENTS_OVRFLW) {
            overflowCount++;
            NRF_RTC1->EVENTS_OVRFLW = 0;
            NRF_RTC1->EVTENCLR      = RTC_EVTEN_OVRFLW_Msk;
            __enable_irq();
        } else { 
            __enable_irq();
            break;
        }
    }
    return ((uint64_t)overflowCount << 24) | counter;
}
```

0xc0170, 2017-06-22:
@pan- Has this been resolved?

pan-, 2017-06-22:
I believe it has been solved.

nvlsianpu, 2017-11-17:
@0xc0170  I was solved. Can you close this?

closed 2017-11-17 by 0xc0170