ARMmbed/mbed-os #4032: Add a `BUTTON` PinName (to targets who have buttons)

janjongboom, 2017-03-27:
### Description

- Type: Enhancement

## Enhancement

A lot of our demo's involve push buttons. Unfortunately there is no set pin name for built-in buttons in the HAL. Thus every demo needs to include a section similar to [this mbed_app.json](https://github.com/ARMmbed/workshop-mbed-connect-asia-lights/blob/master/mbed_app.json#L31) file. It'd be nice to add a `BUTTON1` or `BUTTON` pin name for all devices that have buttons. Preferably combined with a `DEVICE_HAS_BUTTON` macro.

It's a bit of initial work, but I think it will benefit everyone writing applications that have to run on multiple targets.

bulislaw, 2017-03-27:
I agree, the name should be standardized. We are working on new HAL specification, that should be part of it @0xc0170 @c1728p9

bridadan, 2017-05-03:
Closed via #4249 

Good to see this finally resolved 🎉

closed 2017-05-03 by bridadan