ARMmbed/mbed-os #4033: Online IDE not showing build information

andcor02, 2017-03-27:
Has been flagged by several users that the Online-IDE no longer shows the  usage of Ram, Flash & Rom used after compiling.

Can confirm this, anyone have any insight here?

Threads:

https://developer.mbed.org/questions/77527/Compiler-build-informaition/

https://developer.mbed.org/questions/77552/Why-do-the-build-details-show-that-my-pr/

bulislaw, 2017-03-27:
@theotherjimmy

theotherjimmy, 2017-03-27:
So, no target information, no project information, no 2 vs. 5 information.

theotherjimmy, 2017-03-27:
Cannot reproduce.

vargham, 2017-04-01:
I can reproduce. Online IDE with any target. (I tried with STM32, LPC, NXP)
[image]

alejandrosklar, 2017-04-07:
I can reproduce as well. For me, blinky details show up after successful compilation, but details do not show up after successful compilation of a much more complicated and larger program. See below:

Building for STM32 Nucleo L073RZ

[image]
 
 
[image]

theotherjimmy, 2017-04-07:
Thanks for even more test cases on this one guys. I'm trying to narrow down what properties of a build cause it to miss the memory information. As you have pointed out, it seems pretty random, but is very reproducible.

theotherjimmy, 2017-04-26:
I found the bug. It was a typo. My bad. The fix should be deployed soon.

theotherjimmy, 2017-05-03:
solution deployed. Thanks for reporting guys!

closed 2017-05-03 by theotherjimmy