ARMmbed/mbed-os #4034: Not possible to set socket options through socket API

karsev, 2017-03-27:
### Description
- Type: Bug
- Major

---------------------------------------------------------------
## Bug

**Target**
Any 

**Toolchain:**
Any

**Expected behavior**
Should be possible to set socket options for nanostack applications (Thread, 6LoWPAN).
Socket.h provides the setsockopt(int level, int optname, const void *optval, unsigned optlen) function, but does not provide any option names (keys) to choose. Currently the keys for nanostack are defined in the internal API.

**Actual behavior**
Not possible to set options by using the public API. 

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
Currently not possible to implement multicast applications as the default hop limit is 1. 

**Suggested enhancement**
- Define the names (keys) in the public API.
- Update the mbed documentation accordingly.

-----------------------------------------------------------------

## Question

**How to define a hop limit by using an internal API**
For example to set the hop limit ->
my_socket->setsockopt(SOCKET_IPPROTO_IPV6, SOCKET_IPV6_MULTICAST_HOPS, &hops, sizeof(hops));
Add include ->
#include "nanostack/socket_api.h"

bulislaw, 2017-03-27:
@geky @kjbracey-arm

kjbracey-arm, 2017-03-27:
You're using the API in the way it was intended to be used. The bypass mechanism is explicitly provided to let you get at stack-specific socket options which might not be in the stack-independent API.

If your code knows it's running on Nanostack, then what you're doing is fine and legal.

However, in this case, this is an option which may be worth adding to the stack-independent.

I've checked, and lwIP does have a way of setting multicast TTL (although it doesn't have separate IPv4 and IPv6 options). It does default to the same as unicast (ie 255) rather than 1 though, which is unconventional.

kjbracey-arm, 2017-03-27:
Oh, you say "internal API", but those socket option names /are/ in Nanostack's public API.  You're passing options from one stack's public API into the abstraction API, using your knowledge of which stack you're using, and that stack's own documentation and header file.

Far from ideal from an abstraction point of view, but it is using public APIs.

geky, 2017-03-27:
Feel free to create a pr to add any socket options you think should have a common identifier.

I'm not sure it's possible to create an exhaustive list of network options, so allowing stack-specific options lets specific network stacks support features without requiring the features from other network stacks.

karsev, 2017-03-30:
At least we could  add a pointer to the nanostack API where to find the options.

geky, 2017-03-30:
I'm not sure the general API should point to nanostack specifically.

If a user should be able to discover a network option without knowing about the nanostack api beforehand, the network option should probably have an identifier with documentation in the nsapi layer.

cmonr, 2018-01-23:
Considering this resolved as no activity has occurred in a while.

closed 2018-01-23 by cmonr

kjbracey-arm, 2018-01-24:
This has actually just been fully resolved, I think - people have been able to use native Nanostack options for some time, as discussed above.

For the multicast membership that they specifically were after, this was added to the generic NSAPI by #5196, and the Nanostack implementation of that API has just been implemented by #5870.