ARMmbed/mbed-os #4035: Add a `BUILTIN_LED_ON` macro

janjongboom, 2017-03-27:
### Description
- Type: Enhancement

## Enhancement

We have standardized on `LED1` / `LED2` / etc. pin names for the built-in LEDs. Unfortunately these LEDs behave different on different boards. E.g. on K64F you need to set the pin high to turn the LED off, but on  NUCLEO boards you set the pin low to turn the LED off. This is not a problem for blinky, but it is very annoying when writing cross-target applications. People can [macro it up themselves](https://github.com/ARMmbed/simple-cloud-client-example/blob/master/mbed_app.json#L60) but this is always a non-complete list.

I'd suggest we add a `BUILTIN_LED_ON` / `BUILTIN_LED_OFF` macro to the HAL that indicates whether you need to write high or low to a pin to turn the LED on. The downside is that we would need to test on all boards.

bulislaw, 2017-03-27:
That sounds good, but I think we should make it more generic like `HIGH` `LOW` or something similar, as it's really about how the GPIO works.

ChiefBureaucraticOfficer, 2017-10-27:
GitHib issue review: Closed due to inactivity.  Please re-file if critical issues found.

closed 2017-10-27 by ChiefBureaucraticOfficer

janjongboom, 2017-10-30:
@ChiefBureaucraticOfficer I don't agree with closing this issue. It's still an issue.