ARMmbed/mbed-os #4038: No build config in gnuarmeclipse exported project

CarloMara, 2017-03-27:
**Small Info**
- Type: Bug 
- Priority: Blocker 

**Target**
K64F|NUCLEO_F303K8

**Toolchain:**
GCC_ARM

**Toolchain version:**
5.4.0

I'm trying to use the gnuarmeclipse exporter, without any luck so far.
These are the steps that I have taken:

 ```
1. Get a simple mbed project using `mbed import mbed-os-example-blinky`
2. Replace mbed-os with a symlink to my development copy of mbed. I have done this in order to be able to use the master head, with the fixes on the exporters
3. a) Invoke the exporter using `mbed export -m K64F -i gnuarmeclipse`
    b) To make sure that everything is working correctly, I checked in the project folder that the files `.cproject` and `.project` are there.
4. Opened eclipse, choose from the `File` --> `Import`
5. When the import menu pops up, the option `Existing code as Makefile Project` is chosen
6. I than proceed to choose the path of the project, and to select the from the `Tool-chain indexer Settings` the option `Cross ARM GCC`
7. Eclipse than imports the project, and the CDT Indexer fires up.
8. From my understanding, when I click on the hammer icon Eclipse should build the project, and give me a binary file for my uC. That for some reasons it's not happening, even if in the `Console` tab the build is reported as finished (without implying any errors).
If I try to launch the simulator, by pressing the play button, it complains about the fact that it can't find a binary. The `BUILD` folder is empty.
```

In addition to these steps, at the bottom of #3561 @ilg-ul suggested that this issue could be related to the build profiles, that after a check I found to be missing.

I created a new issue because in #3561 this was going OT.

Thanks again,
Carlo

ilg-ul, 2017-03-27:
> When the import menu pops up, the option `Existing code as Makefile Project` is chosen

the exporter generates managed projects, not make projects.

CarloMara, 2017-03-28:
Now I know the difference.
For reference, in order to import the project I went trough the following steps:

`File` --> `Open Project from file System` --> In the popup, I unchecked the option `Search for nested projects` --> Choose the project path --> hit finish.

If this is the way to do it, I can add a step by step guide on the exporters page.

Thanks again,
Carlo

ilg-ul, 2017-03-28:
can you confirm that by properly importing the project, the build configurations were present and the build was successful for all of them?

> I can add a step by step guide on the exporters page

sure.

however, in my opinion, the documentation pages are not very fortunate, there is information about exporters in two places, one is the [Third party development tools](https://docs.mbed.com/docs/mbed-os-handbook/en/latest/dev_tools/third_party/) and one is [Advanced tutorials -> Exporters](https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/exporters/) and there are no links between them.

the first location is too deep, and the names of the intermediate pages are not very clear; for example the **Third party development tools** is not enticing enough to click it and reach the exporters.

the second page does not contain any 'Advanced' information at all, it just lists some items, with very few details.

my suggestion is to rename **Third party development tools** -> **Exporters**, and bring the information from the **Advanced tutorials** -> **Exporters** page here, and only then add a longer, step by step page.

ilg-ul, 2017-03-28:
BTW, I am curious what made you choose to select the project as a makefile project? the small section in the **Third party development tools** page starts exactly by explaining that the exporter creates projects **without** make files.

if that explanation is not easy to understand, we need to improve it.

CarloMara, 2017-03-28:
As you said there are two pages regarding exporters , and that IMO is rather confusing. I have read both, but I must have missed it. Sorry. We should find a definitive way on how to document exporters, in order to avoid issues like this one in the future.

I can confirm that I was able to import the projects following the steps mentioned above. Can you also confirm that that's the best way to do it? 

FYI I was able to build the blink project only on the nucleo, while the build failed with the k64f (see #4044)

See ya,
Carlo

ilg-ul, 2017-03-28:
> I can confirm that I was able to import the projects

ok, great. 

in this case, please update the issue title, since it is not correct. I would suggest 'Importing gnuarmeclipse projects as makefile fails'.

and the 'bug' label attached to this issue should be changed too.

> ... Can you also confirm that that's the best way to do it?

I don't know exactly the steps you did, but I guess it is _the best_, since there shouldn't be more than one ;-)

> the build failed with the k64f

hmmm... something changed in the mean time and again the validation bots did not catch the problem... :-(

ilg-ul, 2017-03-28:
> As you said there are two pages regarding exporters , and that IMO is rather confusing.

@AnotherButler, how can we proceed with an update for the exporter pages?

AnotherButler, 2017-03-28:
@ilg-ul Please feel free to create a PR against whichever of the two pages seems most relevant. Once any missing information is added, we can look into combining the existing information if that seems reasonable.

ilg-ul, 2017-12-16:
@ciarmcom, does this internal ref have any public relevance?

0xc0170, 2017-12-18:
> @ciarmcom, does this internal ref have any public relevance?

It does not, it's for internal use only.

closed 2018-03-28 by ChiefBureaucraticOfficer