ARMmbed/mbed-os #4040: Move ATParser into core mbed OS

janjongboom, 2017-03-27:
### Description
- Type: Enhancement

## Enhancement

We're using the [ATParser](https://github.com/ARMmbed/ATParser) library in more and more places like ESP8266-driver, MTSAS cell driver; and especially with our move to modules I expect this to increase even more, as they probably rely on AT commands for communication. Would it be an idea to move the ATParser library into core OS, rather than as a separate library?

/cc @sarahmarshy

sarahmarshy, 2017-03-27:
+1 @geky

geky, 2017-03-27:
I don't really have an opinion, I'd go for it if it's easier for users.

cc @sg-

sg-, 2017-03-28:
If there are going to be multiple dependencies this seems reasonable but probably worth waiting or seeing what @kjbracey-arm and @hasnainvirk are up to with cellular work.

kjbracey-arm, 2017-03-28:
Yes, this is going to happen as part of the cellular stuff.

There will need to be some slight initialisation changes to use the updated ATParser as it is being revised to work with any device, rather than being locked to serial.

Although maybe transparent backwards compatibility could be achieved - we'll have to investigate.

hasnainvirk, 2017-03-28:
Code can be checked here. We are in the intensive testing phase right now. 
https://github.com/hasnainvirk/mbed-os/tree/cellular_feature_br/
Soon we will begin merging to mbed-os for 5.5 dev cycle

sg-, 2017-03-28:
I really wish we'd all work on feature branches in mbed-os tree 👍

janjongboom, 2017-03-28:
@sg- I for one do not have push access to ARMmbed/mbed-os. Maybe we should start with allowing writes to feature branches for all mbed Staff.

hasnainvirk, 2017-03-28:
@sg- :) I don't have push access to mbed-os so I have my own fake feature branch :D

sg-, 2017-03-28:
That is what PRs are for :)

cmonr, 2018-01-23:
Closing due to merge.

closed 2018-01-23 by cmonr