ARMmbed/mbed-os #4052: NTP support

chrissnow, 2017-03-28:
## Enhancement

It would be great if the Ethernet stack was extended to include NTP (or SNTP)
LWIP already has some support for it.

Ideally it would be as simple as declaring an instance of NTP and giving it a time server
it would then keep the RTC(if there is one) in sync and then calls to time() would work.

0xc0170, 2017-03-29:
cc @kjbracey-arm @geky

geky, 2017-03-29:
I agree this would be a nice feature to have, I don't know of anyone working on it at the moment.

chrissnow, 2017-04-25:
I've modified this
https://developer.mbed.org/users/donatien/code/NTPClient/
and moved it to github here.
https://github.com/chrissnow/mbed-NTP

it uses the new API's and the basic functionality works.

a simple example is
```c++
NTPClient ntp = NTPClient(&eth);

if (ntp.setTime("pool.ntp.org") == NTP_OK) {
	printf("Clock set by NTP\r\n");
} else 
{
	printf("NTP failed\r\n");
}
```

I am working on making an option of it creating a timekeeping thread but that's unfinished.

pauluap, 2018-06-04:
I'm not sure if this is a known thing, but LWIP has a sntp app, and I was able to use it unmodified after enabling the necessary defines and configuring the function to set the time.

https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_LWIP/lwip-interface/lwip/src/apps/sntp/lwip_sntp.c

closed 2019-10-31 by ciarmcom