ARMmbed/mbed-os #4053: Windows 10 @ NXP KLXX Platforms

andcor02, 2017-03-28:
This is a somewhat strange issue that seems to be affecting a group of users. 

**The problem:**

Some of the NXP KLXX series (KL25Z, KL46Z.. etc) are still shipped with outdated firmware that is not recognised by Windows 8/10. Therefore will not show the "MBED" drive. Ontop of this, you cannot update this old firmware in Windows 10. 

See here for examples of the problem:

- https://developer.mbed.org/questions/77014/Reset-to-factory-settings/
- https://developer.mbed.org/questions/77561/How-to-i-connect-FRDM-KL25Z-to-windows-1/?c=26921
- https://mcuoneclipse.com/2016/08/01/bricking_and_recovering_opensda_boards_in_windows_8_and_10/

**The hacked solutions,** 3 differing ones flagged here :https://developer.mbed.org/questions/77014/Reset-to-factory-settings/

1. Run the firmware update using a Windows 7 PC (or earlier).
2. Disable library search using the group policy editor in Windows 10 pro (and run a firmware update)
3. Disable library search using the register editor in Windows 10 (and run a firmware update)

Regards,

Andrea, team mbed

0xc0170, 2017-03-29:
@andcor02 What is your proposal to solve this issues?

sg-, 2017-12-16:
For boards that run DAPLink bootloaders there is a blog on how to update them using windows 10. These instructions would work for other bootloaders too making it safe to update using MSC and windows 10 if you replaced the DAPLink bootloader with an alternative. https://os.mbed.com/blog/entry/DAPLink-bootloader-update/

closed 2017-12-16 by sg-