ARMmbed/mbed-os #4055: mbed test is not working in mbed-os-example-bootloader application

tkaman, 2017-03-28:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Related issue: `#abc`
- Priority: Minor

I imported mbed-os-example-bootloader application and tried to run mbed test with it, but it is not working. 

$ mbed test --list
Traceback (most recent call last):
  File "C:\ARM\SourceCode\mbed-os-example-bootloader\mbed-os\tools\test.py", line 128, in <module>
    app_config=options.app_config))
  File "C:\ARM\SourceCode\mbed-os-example-bootloader\mbed-os\tools\test_api.py", line 2017, in find_tests
    silent=True, app_config=app_config)
  File "C:\ARM\SourceCode\mbed-os-example-bootloader\mbed-os\tools\build_api.py", line 339, in prepare_toolchain
    add_regions_to_profile(build_profile, config, cur_tc)
  File "C:\ARM\SourceCode\mbed-os-example-bootloader\mbed-os\tools\build_api.py", line 290, in add_regions_to_profile
    profile["common"].append("-D%s=0x%x" %  define)
TypeError: 'NoneType' object has no attribute '__getitem__'
[ERROR] 'NoneType' object has no attribute '__getitem__'
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\ARM\SourceCode\mbed-os-example-bootloader\mbed-os\tools\test.py -t ARM -m K64F --source C:\ARM\SourceCode\mbed-os-example-bootloader --build C:\ARM\SourceCode\mbed-os-example-bootloader\BUILD\tests\K64F\ARM --test-spec C:\ARM\SourceCode\mbed-os-example-bootloader\BUILD\tests\K64F\ARM\test_spec.json --list" in "C:\ARM\SourceCode\mbed-os-example-bootloader"

I did not touch any files in the repo.

tamkam01@E109896 MINGW64 /c/ARM/SourceCode/mbed-os-example-bootloader (master)
$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working tree clean

tamkam01@E109896 MINGW64 /c/ARM/SourceCode/mbed-os-example-bootloader/mbed-os (mbed-os-5.4)
$ git status
On branch mbed-os-5.4
Your branch is up-to-date with 'origin/mbed-os-5.4'.
nothing to commit, working tree clean

mbed cli version
tamkam01@E109896 MINGW64 /c/ARM/SourceCode/mbed-os-example-bootloader (master)
$ mbed --version
1.0.0

mbed os is the latest version
tamkam01@E109896 MINGW64 /c/ARM/SourceCode/mbed-os-example-bootloader/mbed-os (mbed-os-5.4)
$ git pull
Already up-to-date.

tamkam01@E109896 MINGW64 /c/ARM/SourceCode/mbed-os-example-bootloader/mbed-os (mbed-os-5.4)
$ git describe
latest

On mbed os master branch the same issue is seen.

Strange thing is that another example application (mbed-os-example-blinky) with the same mbed os version is working fine

---------------------------------------------------------------
## Bug

**Target**
without target, mbed test --list also fails

**Toolchain:**
not related

**Toolchain version:**
-
**mbed-cli version:** 
1.0.0

**meed-os sha:**
$ git log -n1 --oneline
f4864dc64 Merge pull request #3930 from ARMmbed/release-candidate

**DAPLink version:**
-
**Expected behavior**

**Actual behavior**

**Steps to reproduce**
import mbed-os-example-bootloader
cd mbed-os-example-bootloader
mbed test --list

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2017-03-28:
cc @theotherjimmy

theotherjimmy, 2017-03-28:
huh. I'll look into this in more depth.

0xc0170, 2017-03-29:
@theotherjimmy any findings?

theotherjimmy, 2017-03-29:
Not yet.

theotherjimmy, 2017-03-29:
I figured out what the problem was. #4070 has the solution.

PR #4070 (2017-04-06): Elide adding regions to profile when empty
Resolves #4055

The add_regions_to_profile function assumed that there would always be a
profile to add regions to. If the profile is not specified, we would get
a traceback. Instead of bombing horribly, we should just elide adding
the region list to the profile when it does not exist.

# Testing
- [x] manual verification that the issue in #4055 no longer happens

# CC
@0xc0170 
@tkaman
closed 2017-04-06 by sg-