ARMmbed/mbed-os #4057: "device_name" not found - attempting to compile mbed-os-example-bootloader on NCS36510

jacobjohnson-ON, 2017-03-28:
### Description
- Type: Bug
- Related issue: `#bootloader`
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NCS36510

**Toolchain:**
GCC_ARM

**Toolchain version:**
5.06
**mbed-cli version:** 
1.0.0

**meed-os sha:**
8b Merge pull request #8 from 0xc0170/fix_readme_blinky

**DAPLink version:**
0240

**Expected behavior**
Expected to compile and run the HAL and FlashIAP tests.  

**Actual behavior**

C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootloader>mbed compile -v
[mbed] Working path "C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootloader
" (program)
[mbed] Exec "python -u C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootload
er\mbed-os\tools\make.py -t GCC_ARM -m NCS36510 --source . --build .\BUILD\NCS36510\GCC_ARM -v" in C
:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootloader
Traceback (most recent call last):
  File "C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootloader\mbed-os\tool
s\make.py", line 285, in <module>
    toolchain))
  File "C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootloader\mbed-os\tool
s\build_api.py", line 481, in build_project
    build_profile=build_profile)
  File "C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootloader\mbed-os\tool
s\build_api.py", line 339, in prepare_toolchain
    add_regions_to_profile(build_profile, config, cur_tc)
  File "C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootloader\mbed-os\tool
s\build_api.py", line 286, in add_regions_to_profile
    regions = list(config.regions)
  File "C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootloader\mbed-os\tool
s\config.py", line 482, in regions
    cmsis_part = Cache(False, False).index[self.target.device_name]
  File "C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootloader\mbed-os\tool
s\targets.py", line 256, in __getattr__
    result = self.__getattr_helper(attrname)
  File "C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootloader\mbed-os\tool
s\targets.py", line 243, in __getattr_helper
    % (attrname, self.name))
AttributeError: Attribute 'device_name' not found in target 'NCS36510'
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-exampl
e-bootloader\mbed-os\tools\make.py -t GCC_ARM -m NCS36510 --source . --build .\BUILD\NCS36510\GCC_AR
M -v" in "C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootloader"
---

C:\Users\fg8dfg\Documents\MBED_Partner_Workshop\Code\mbed-os-example-bootloader>

**Steps to reproduce**

0xc0170, 2017-03-28:
@theotherjimmy I have seen this today on 2 devices. I cant reproduce this locally. The attribute error somehow gets in, it's not clear to me from the traceback why it needs this symbol for compiling?

theotherjimmy, 2017-03-28:
Ah, I'll catch that exception and throw a better warning. `device_name` should be populated with something that is a CMSIS Part. This is required for bootloader builds to work properly.

theotherjimmy, 2017-03-28:
PR to improve the error messages in a moment.

theotherjimmy, 2017-04-25:
@jacobjohnson-ON the bootloader bulids require a information from CMSIS PACKS. This requirement should be better highlighted in the error messages now that #4061 has been merged. Could you comment on the status of this issue?

ashok-rao, 2017-11-02:
@theotherjimmy .. I saw this today for SiLabs part. Initially, I got an error message (for something I was building with bootloader) that device_name is not present. I then added "device_name": ["EFR32MG12P332F1024GL125"] to targets.json. I got the device's name from KEIL's webpage: http://www.keil.com/dd2/siliconlaboratories/efr32mg12p332f1024gl125/  this is same device name that appears in CMSIS packs in MDK as well .. however, I still cannot build for this target.. some logs are below..

C:\Ashok\SiLabs_ThunderSense2\example-sources-internal>mbed compile -t GCC_ARM -m TB_SENSE_12 --app-config configs\mesh_thread.json -vv
[mbed] Working path "C:\Ashok\SiLabs_ThunderSense2\example-sources-internal" (program)
[mbed] Exec "c:\python27\python.exe -u C:\Ashok\SiLabs_ThunderSense2\example-sources-internal\mbed-os\tools\make.py -t GCC_ARM -m TB_SENSE_12 --source . --build .\BUILD\TB_SENSE_12\GCC_ARM --app-config configs\mesh_thread.json -v" in C:\Ashok\SiLabs_ThunderSense2\example-sources-internal
Traceback (most recent call last):
  File "C:\Ashok\SiLabs_ThunderSense2\example-sources-internal\mbed-os\tools\make.py", line 296, in <module>
    stats_depth=options.stats_depth)
  File "C:\Ashok\SiLabs_ThunderSense2\example-sources-internal\mbed-os\tools\build_api.py", line 509, in build_project
    build_profile=build_profile)
  File "C:\Ashok\SiLabs_ThunderSense2\example-sources-internal\mbed-os\tools\build_api.py", line 355, in prepare_toolchain
    add_regions_to_profile(profile, config, cur_tc)
  File "C:\Ashok\SiLabs_ThunderSense2\example-sources-internal\mbed-os\tools\build_api.py", line 296, in add_regions_to_profile
    regions = list(config.regions)
  File "C:\Ashok\SiLabs_ThunderSense2\example-sources-internal\mbed-os\tools\config\__init__.py", line 504, in regions
    if self.target.device_name not in cache.index:
TypeError: unhashable type: 'list'
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Ashok\SiLabs_ThunderSense2\example-sources-internal\mbed-os\tools\make.py -t GCC_ARM -m TB_SENSE_12 --source . --build .\BUILD\TB_SENSE_12\GCC_ARM --app-config configs\mesh_thread.json -v" in "C:\Ashok\SiLabs_ThunderSense2\example-sources-internal"
---**

ashok-rao, 2017-11-02:
Thanks @theotherjimmy , was able to get past without the [].

theotherjimmy, 2017-11-02:
@ashok-rao Do you think the documentation should be improved to help users prevent this bug in the future?

ashok-rao, 2017-11-02:
Would be nice to have them in the docs. I didn't find any docs _particularly on the keys_ we/every target uses in targets.json .. I also may not have looked in the right places though..!

theotherjimmy, 2017-11-02:
That document is the ["Adding and Configuring Targets section](https://os.mbed.com/docs/v5.6/tools/adding-and-configuring-targets.html). It's somewhat hard to find. Where would you expect to find it?

ashok-rao, 2017-11-12:
@theotherjimmy .. I'd expect to find this somewhere in the porting guide (adding / configuring a new target) under adding a new MCU here: https://os.mbed.com/docs/v5.6/reference/contributing-target.html

theotherjimmy, 2017-11-13:
Thanks @ashok-rao. I'll add a "Bootloader" section there and make sure to add some explanatory text to the "Target Description" section pointing to the link I posted above.

cmonr, 2018-06-14:
Closing since no activity has occured in months.

Feel free to open a new issue if this is still a problem.

PR #4061 (2017-03-29): Improve error messages for bootloader build errors
The error messages related to missing or broken device names in the
targets.json file were bad/just python tracebacks. This PR changes that
behavior, by having the Config object raise ConfigExceptions for those
mis-configurations.

Resolves #4057

## Testing
 - [x] Normal /morph test
 - [x] Manual exercise of each corner case and verification that the
 messages seen are good.

### Manual testing
 - `mbed_app.json`
```
{
    "target_overrides":{
        "NCS36510":{
            "target.restrict_size": "0x40000"
        }
    }
}
```
- `mbed com [... cut]
closed 2018-06-14 by cmonr