ARMmbed/mbed-os #4067: Too high static reservations for HEAP and STACK in most IAR builds

SeppoTakalo, 2017-03-29:
### Description

Most of the IAR build statically define the stack size to be something between 512 bytes to 24 kbytes.
This stack, however, is not even used in RTX case. Or maybe it is only used for interrupts.
Each other threads define their own stack, something between 1kB - 6kB. There is no justification to use 24kB for boot stack.

For example, Ublox EVK Odin W2 defines this:
```
/*-Sizes-*/
/*Heap 1/2 of ram and stack 1/8*/
define symbol __ICFEDIT_size_cstack__ = 0x6000;
define symbol __ICFEDIT_size_heap__   = 0x18000;
```

Which means we have statically allocated 24kB for the stack that is mostly unused, and 98kB for the heap. Total 122880 kB. That leaves only 73 kB for other use (Threads, .data and .bss sections).

This is causing problems when getting anything to fit into these devices. This issues is mostly visible on IAR as it does not dynamically grow the heap.

We should align the stack sizes between boards. Some sensible number, with justification, should be used.

0xc0170, 2017-03-29:
> We should align the stack sizes between boards. Some sensible number, with justification, should be used.

Any proposals ?

@sg- @c1728p9 @geky @kjbracey-arm @pan-

kjbracey-arm, 2017-03-29:
Personally, I think it's more important to have a way for the application to be able to easily configure it via their mbed_app.json.

Only each application knows how much it's intending to do in main().

The amount of stack space needed by an application is (almost totally) a function of the application, not the board.

I would suggest there should be a constant default size across all boards for the main stack, maybe 2-4K. Applications needing more than that should have to configure it higher. And having to configure it up regardless of board is going to be a bit less annoying than having an app which works with the default stack size on one board, but fails on a board with a smaller default stack size.

deepikabhavnani, 2018-07-19:
https://github.com/ARMmbed/mbed-os/pull/7238

deepikabhavnani, 2018-12-28:
this issue will be resolved by https://github.com/ARMmbed/mbed-os/pull/9092

closed 2019-02-05 by ciarmcom