ARMmbed/mbed-os #4068: unstability with ethernet UDP connection, not detected by mbed tests on NUCLEO_F746NG

adustm, 2017-03-29:
### Description
- Type: Bug + Enhancement
- Related issue: https://developer.mbed.org/questions/77138/EthernetInterface-unreliable-on-NUCLEO-F
- Priority: Blocker 

---------------------------------------------------------------
## Bug
Create an UDP socket on the MCU board. Use an external UDPclient that sends messages every millisedoncs (or close to that). After a few seconds, the MCU will not answer to UDP client anymore.

Use this example code: 
```
#if !FEATURE_LWIP
    #error [NOT_SUPPORTED] LWIP not supported for this target
#endif

#include "mbed.h"
#include "EthernetInterface.h"
#include "TCPServer.h"
#include "UDPSocket.h"

#define PORT 7
int main() {
    EthernetInterface eth;
    SocketAddress client;
    char buffer[256];
 
    eth.connect(); 
    UDPSocket server;
    server.open(&eth);
 
    while ( server.bind(PORT) != 0);  
    
    while (true) {     
        int n = server.recvfrom(&client, buffer, sizeof(buffer));       
        if (n>0) {   
          if (!strncmp(buffer,"id\n",3)) {
            server.sendto(client, "hello\n", 6);
          } else {
            server.sendto(client, "unknown\n", 8);
          }      
       }
    }
}
```
**Target**
NUCLEO_F746ZG

**Toolchain:**
ARM (other not tested)

**Expected behavior**
The F746ZG should be able to keep working, even if it is missing some messages.

**Actual behavior**
Once the F746 stops answering, it never work anymore until a RESET. I have exported the test in a debugger, it is not crashing, it is only in idle... so it is difficult to me to know what happened.
I have made a lot of trials but I don't know how to debug that now... any help is welcome

**Steps to reproduce**
cf above code example + above link on developper.mbed.org

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
mbed-os tests have passed on this platform and did not detect any issue
the MCU is not really crashed somewhere, it is in idle -> how can I debug the threads ?

**Suggested enhancement**
Could you add a new test for that ?
Could you explain how to debug threads ?

0xc0170, 2017-03-29:
cc @geky

geky, 2017-06-12:
Curious, does the udp echo test not fail?
https://github.com/ARMmbed/mbed-os/tree/master/features/FEATURE_LWIP/TESTS/mbedmicro-net/udp_echo/main.cpp

We do have more intensive testing, but we haven't been able to turn them on yet:
https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_LWIP/TESTS/mbedmicro-net/udp_packet_pressure/main.cpp

You can try running these with the following mbed commands:
```
mbed test --compile -t NUCLEO_F746NG -m ARM -DMBED_EXTENDED_TESTS -n features-feature_lwip-tests-mbedmicro-net-udp_echo,features-feature_lwip-tests-mbedmicro-net-udp_packet_pressure
mbed test --run -t NUCLEO_F746NG -m ARM -n features-feature_lwip-tests-mbedmicro-net-udp_echo,features-feature_lwip-tests-mbedmicro-net-udp_packet_pressure
```

danielklioc, 2017-11-14:
@adustm did you find out what was the problem with UDP socket stopping working with high packet load ? I have similar problem with NucleoF746/767.

cmonr, 2018-06-14:
Closing since no activity has occured in months.

Feel free to open a new issue if this is still a problem.

closed 2018-06-14 by cmonr