ARMmbed/mbed-os #4075: Missing mbed_config.h from exported project

vargham, 2017-03-30:
### Description
- Type: Bug
- Priority: Major

There is no mbed_config.h in exported project. Tried with STM32F103, STM32F429, LPC1768 mbed and mbed-os lib also. Tried with EmBlocks, GCC make and SW4STM32. Seems all targets and IDEs are affected.

```
$ make
Compile: main.cpp
cc1plus.exe: fatal error: mbed_config.h: No such file or directory
compilation terminated.
make[1]: *** [/cygdrive/d/_dev/_tmp/arm_tests/mbed_blinky/Makefile:241: main.o] Error 1
make: *** [Makefile:26: all] Error 2
```
---------------------------------------------------------------
## Bug

**Target**
STM32F103|STM32F429|LPC1768|Maybe others

**Toolchain:**
Online exporter

**Toolchain version:**
Current online exporter (2017-03-30)

**Expected behavior**
Exported project should contain all files.

**Actual behavior**
Build fails, missing mbed_config.h

**Steps to reproduce**
Create project "mbed_blinky" for LPC1768 and export it to offline GCC makefile. Build will fail because missing mbed_config.h

bulislaw, 2017-03-30:
@theotherjimmy

0xc0170, 2017-03-30:
I've just hit something related, that I had to use old project.py script to test something for mbed 2 and this is traceback:

```
python tools\project.py -m LPC1768 -n MBED_A2 -i uvision
Scan: file
Scan: env
Scan: hal
Scan: targets
Traceback (most recent call last):
  File "tools\project.py", line 247, in <module>
    main()
  File "tools\project.py", line 243, in main
    build_profile=profile)
  File "tools\project.py", line 93, in export
    build_profile=build_profile, silent=silent)
  File "C:\Code\git_repo\github\mbed-official\tools\export\__init__.py", line 327, in export_project
    macros=macros)
  File "C:\Code\git_repo\github\mbed-official\tools\export\__init__.py", line 194, in generate_project_files
    exporter.generate()
  File "C:\Code\git_repo\github\mbed-official\tools\export\uvision\__init__.py", line 204, in generate
    'project_files': sorted(list(self.format_src(srcs).iteritems()),
  File "C:\Code\git_repo\github\mbed-official\tools\export\uvision\__init__.py", line 175, in format_src
    grouped = self.group_project_files(srcs)
  File "C:\Code\git_repo\github\mbed-official\tools\export\exporters.py", line 150, in group_project_files
    data = sorted(sources, key=self.make_key)
  File "C:\Code\git_repo\github\mbed-official\tools\export\exporters.py", line 134, in make_key
    rel_path = relpath(src, self.resources.file_basepath[src])
KeyError: 'mbed_config.h'
```

(missing mbed_config ?)

vargham, 2017-03-30:
Workaround: Copy mbed_config.h from previous export. (This bug was not present before last week.)

theotherjimmy, 2017-03-30:
@0xc0170 that's the error that #4013 Fixes. @vargham I'll look into the online problem.

closed 2017-12-16 by sg-