ARMmbed/mbed-os #4076: Reduce visual clutter in top-level directory

wdwalker, 2017-03-30:
### Description
- Type: Enhancement
- Priority: Minor

## Enhancement

**Reason to enhance or problem with existing solution**

When I use the mbed CLI to work with mbed OS 5.4, I end up with a number of files in the top level directory that are for mbed itself: mbed-os.lib, mbed-os, mbed_settings.py[c], a directory and *.lib pair for each library I import, mbed_app.json for my app, etc. In addition, there are other hidden files, such as .mbed and .mbedignore.

It would be great to reduce this clutter.

**Suggested enhancement**

Make a hidden subdirectory to hold the mbed configs. I don't have this well thought out -- were it not for the existence of the .mbed file, I'd suggest making a .mbed directory to hold all of the mbed configs, which might be considered the mbed equivalent of the .git directory.

I've marked this as "minor" because the impact is mostly related to a minor facet of developer experience and has no impact on the functionality of an application. I will not be offended at all if this enhancement request is rejected or closed as "won't fix".

**Pros**

Visual clutter in the top-level directory is reduced.

**Cons**

Hiding things can lead to confusion. This may also require a large change to the mbed CLI just for the benefit of reducing clutter.

bulislaw, 2017-03-30:
That should probably go to https://github.com/ARMmbed/mbed-cli

wdwalker, 2017-03-30:
Ah - true. Sorry. Will move.

closed 2017-03-30 by wdwalker